chipsalliance/rocket-chip #2366: Should clockGate be a member or CoreParams

davidmlw, 2020-03-26:
Two reasons for this:
1. clockGate is an attribute of every core, or every tile.
2. for rocket core, FrontendModule depends on HasRocketCoreParameters to get clockGate attribute, making FrontedModule a specific core dependent design. if clockGate is a global attribute of CoreParams, then FrontenModule will be a generic design to any core.
https://github.com/chipsalliance/rocket-chip/blob/2f6c74990dceb484ec032d8c31ff86fec8ec5a6e/src/main/scala/rocket/Frontend.scala#L91

aswaterman, 2020-03-26:
FrontendModule is not especially generic. It’s microarchitecturally coupled
to the Rocket design. In my experience, trying to make these large core
modules generic has been unproductive—see the FPU, for example. Making
smaller modules more generic has been somewhat more successful.

I’m fine with making clockGate more generic.

davidmlw, 2020-03-27:
I agree your insight. Chisel is very cool to make the small module generic and even make them in language itself.

Back to this specific rocket.FrontendModule, it is make some sense to make clockGate in CoreParams, so FrontendModule can be decoupled with rocket.

I will file a PR to master to make clockGate more generic.