chipsalliance/rocket-chip #2371: We should forbid synthesizing simulation library, like a clock-gate simulation model

ultragamza, 2020-03-27:
other enhancement

no functional change

request

**Other information**
-
**If the current behavior is a bug, please provide the steps to reproduce the problem:**
-
**What is the current behavior?**
Accidentally, one of our customers didn't implement EICG_wrapper. 
They can do simulation & synthesis.
I think it could make big trouble someday.
So I'm proposing to forbid synthesizing simulation library which should be implemented for a specific chip environment.

**What is the expected behavior?**
The function of EICG_wrapper, ClockDivider2 and ClockDivider3 are not changed.
But if they are synthesized, you can see the following message.
module 'Error_EICG_wrapper_must_be_ported' not found [".../src/main/resources/vsrc/EICG_wrapper.v":20]

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2020-03-27:
EICG_wrapper is actually synthesizable on FPGA, and I believe some users (including SiFive, IIRC) rely on that fact.  Simply issuing an error if SYNTHESIS is defined will cause problems in that flow.  Does Vivado define some macro like VIVADO so we can make an exception for that tool?

ultragamza, 2020-03-27:
Is there any macro to recognize VIVADO which SiFive and the other users are using?

ultragamza, 2020-03-27:
I think we could make separate repositories for this, which has clock-gate, clock-mux, clock-divider, scan clock mux, synchronizer, reset synchronizer,...
Then, we can have several repositories that already have been ported for each tech(ss28,tsmc28lp,...).
**Then physical implementation team doesn't need to do the same job.**
I think it's a better way to resolve this problem and the wake could help us.

ultragamza, 2020-03-27:
Unfortunately, there is no way to detect VIVADO.
https://forums.xilinx.com/t5/Synthesis/vivado-synthesizer-defined-macro/td-p/1076298

aswaterman, 2020-03-27:
That sounds reasonable, but it needs to be done before making this proposed change.

closed 2023-06-08 by scott-kim-7