chipsalliance/rocket-chip #2388: Illegal csr access mismatch between Rocket chip and spike

JaewonHur, 2020-04-02:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: proposal

**Other information**
 

When try to read illegal csr addres '0x7b3'
Rocket chip raise exception, but spike just ignore it.

However when try to read illegal csr address '0x7c1'
Spike raise exception while Rocket chip doesn't.
It seems '0x7c1' is related to chicken bits (?)

By the way, also when Rocket chip tries to execute 'csrrw zero,0x7b3,s6',
It raises exception, Is it okay to raise exception even if the destination register is zero?

I'm using spike commit : 157143b5038ef2b92b1c70d445b566d57b6f5096
and Rocket : 9a1da4e01e3e767451d9ee4a4f0ee8a55a3313da

Here are tests.
[test_7b3.zip](https://github.com/chipsalliance/rocket-chip/files/4419371/test_7b3.zip)
[test_7c1.zip](https://github.com/chipsalliance/rocket-chip/files/4419372/test_7c1.zip)

aswaterman, 2020-04-02:
Not all implementations provide the same CSRs.

0x7b3 is an optional debug register, which spike provides but rocket does not.

0x7c1 is a custom rocket-specific register, which spike does not implement.

JaewonHur, 2020-04-02:
Ah, Thanks for fast reply.

closed 2020-04-02 by JaewonHur