chipsalliance/rocket-chip #2402: Bug in RVC expansion of reserved instructions

tommythorn, 2020-04-07:
This was fun to find (though I have not reproduced it in Chisel, this was hard enough already).

According to Tabel 16.6, page 113 in the current RISC-V spec, these
are reserved compressed instructions (and thus should trap):
```
101  1  11  -   10  -  01
101  1  11  -   11  -  01
```

However RVCDecoder expands these to `SLTW` and `SLTUW` respectively, instead of `unimp`.

The troubled section is
https://github.com/chipsalliance/rocket-chip/blob/d76703884d9dbd2965536a28a8499928076b89d5/src/main/scala/rocket/RVC.scala#L103
which is unchanged from the original commit (but was moved three times): 
```
commit 058396aefe4103046406be63e1a4c0178c4d13b3
Author: Andrew Waterman <waterman@cs.berkeley.edu>
Date:   Fri Jul 29 16:36:07 2016 -0700

    [rocket] Implement RVC
```

tommythorn, 2020-04-07:
Pinging @aswaterman

aswaterman, 2020-04-07:
SLTW and SLTUW aren't instructions, though.  If you mean these opcodes expand to what _would_ be SLTW and SLTUW if we defined W versions of those, then there's no bug, because the only goal is to raise an illegal instruction exception.  Or did you mean something else...?

tommythorn, 2020-04-07:
I see.  Provides disassembly for these, which I guess is then a different bug.  Still, if all you intended was to provide an illegal instruction, why did you use 2.U, 3.U specifically (instead of eg. 2.U, 2.U)?

In quadrant 0, the reserved instructions are explicitly translated to unimp so this is certainly surprising.

closed 2020-04-07 by tommythorn

aswaterman, 2020-04-07:
Sorry for the confusing aspect of this API.  I think the 2/3 came from symmetry with 6/7 earlier in the list, to simplify decode.

tommythorn, 2020-04-07:
It's fine, not broken, but it feels a bit fragile; if those opcodes were defined in future (but RVC not revised) then this would silently turn into a bug.

aswaterman, 2020-04-07:
Or an unintended feature :) but yeah.