chipsalliance/rocket-chip #2404: Option WithL1DCacheWays not working

kyaso, 2020-04-07:
**Type of issue**: bug report

**Impact**: unknown

**Other information**
 
We tried using the `WithL1DCacheWays` option to get a DCache with 8 ways, but the size of DCache does not change and stays at 16KiB. 

**However**, when manually setting `nWays=8` for `dcache` inside `WithNBigCores`, the size of DCache is correct.

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

1. Create a config in _system/Configs.scala_ using `WithL1DCacheWays` options:
```
class DefaultFPGAConfig extends Config(new WithNBigCores(1) ++ new WithL1DCacheWays(8) ++ new BaseFPGAConfig
```
2. Go to vsim directory and compile:
```
make -j4 CONFIG=freechips.rocketchip.system.DefaultFPGAConfig
```

**What is the current behavior?**
DCache size = 16KiB
(see generated _dts_ and _behav_srams_)

**What is the expected behavior?**
DCache size = 32KiB

**Please tell us about your environment:**

Git commit: d76703884d9dbd2965536a28a8499928076b89d5
sbt version: 1.3.3
OS: `Linux crampus 3.10.0-957.21.2.el7.x86_64 #1 SMP Wed Jun 5 14:26:44 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux`

jackkoenig, 2020-04-08:
`WithNBigCores` constructs the `DCacheParams` overriding the parameters to the right (there's no `site` or `up` lookup for the existing parameterization). You need to have `WithL1DCacheWays` to the left so that it overrides the `DCacheParams` from `WithNBigCores`) try:
```scala
class DefaultFPGAConfig extends Config(new WithL1DCacheWays(8) ++ new WithNBigCores(1) ++ new BaseFPGAConfig)
```
Note that I haven't actually run this, so there could be other issues, but that will ensure the numbers of ways from `WithL1DCacheWays` is the final value.

kyaso, 2020-04-08:
Yes, putting `WithL1DCacheWays(8)` in front of `WithNBigCores` results in the correct number of ways.

closed 2020-04-08 by kyaso