chipsalliance/rocket-chip #2405: Is this dcache_miss triggered replay_ex valid?

davidmlw, 2020-04-08:
https://github.com/chipsalliance/rocket-chip/blob/d76703884d9dbd2965536a28a8499928076b89d5/src/main/scala/rocket/RocketCore.scala#L481-L486

`io.dmem.resp.valid` combinatially propogates to `dcache_miss` and then propogates to `replay_ex`, and `replay_ex` sequentially propogates to repaly_wb and finally an instruction replay is triggered.

my questions is that `dcache_miss` is a writeback stage known signal, when dcache_miss is known, the instruction in current execute stage is not the one should be replayed.

Another observation is replay_wb depends on io.dmem.s2_nack, and will trigger current writeback stage instruction replay, which is right.
https://github.com/chipsalliance/rocket-chip/blob/d76703884d9dbd2965536a28a8499928076b89d5/src/main/scala/rocket/RocketCore.scala#L627-L629

This brings another question, that, what is the relationship of dcache_miss is asserted as dmem.resp.valid does not show two cycles later after dmem.req.valid; and dmem.s2_nack is also a two cycle later signal. So what is the relationship of dmem.resp.valid and dmem.s2_nack?

Is this assertion valid, assert(!resp.valid || s2_nack after two cycles of req.valid)?

aswaterman, 2020-04-09:
io.dmem.resp.valid and io.dmem.s2_nack are mutually exclusive.

davidmlw, 2020-04-10:
I see. Let me try to use code:
`assert((io.dmem.resp.valid ^ io.dmem.s2_nack) && delay(io.dmem.req.valid, 2 cycles))`
And the instruction is either commited when assert(resp.valid), or will be replayed when assert(s2_nack).

closed 2022-04-12 by michael-etzkorn