chipsalliance/rocket-chip #2406: How to simulate rocketchip without SimDTM and disable debug module?

zhh124, 2020-04-08:
Currently I emulate rocketchip successfully with verilator, when I disable SimDTM module, and simulate with iverilog, not got correct result. the CONFIG is DefaultConfig when generate RTL.

**disable SimDTM:**
/*  SimDTM SimDTM ( // @[Periphery.scala 232:25:freechips.rocketchip.system.DefaultConfig.fir@207733.4]
    .clk(SimDTM_clk),
    .reset(SimDTM_reset),
    .debug_req_ready(SimDTM_debug_req_ready),
    .debug_req_valid(SimDTM_debug_req_valid),
    .debug_req_bits_addr(SimDTM_debug_req_bits_addr),
    .debug_req_bits_data(SimDTM_debug_req_bits_data),
    .debug_req_bits_op(SimDTM_debug_req_bits_op),
    .debug_resp_ready(SimDTM_debug_resp_ready),
    .debug_resp_valid(SimDTM_debug_resp_valid),
    .debug_resp_bits_data(SimDTM_debug_resp_bits_data),
    .debug_resp_bits_resp(SimDTM_debug_resp_bits_resp),
    .exit(SimDTM_exit)
  );*/

/*  assign ldut_debug_clockeddmi_dmi_req_valid = SimDTM_debug_req_valid; // @[Periphery.scala 166:15:freechips.rocketchip.system.DefaultConfig.fir@207740.4]
  assign ldut_debug_clockeddmi_dmi_req_bits_addr = SimDTM_debug_req_bits_addr; // @[Periphery.scala 166:15:freechips.rocketchip.system.DefaultConfig.fir@207740.4]
  assign ldut_debug_clockeddmi_dmi_req_bits_data = SimDTM_debug_req_bits_data; // @[Periphery.scala 166:15:freechips.rocketchip.system.DefaultConfig.fir@207740.4]
  assign ldut_debug_clockeddmi_dmi_req_bits_op = SimDTM_debug_req_bits_op; // @[Periphery.scala 166:15:freechips.rocketchip.system.DefaultConfig.fir@207740.4]
  assign ldut_debug_clockeddmi_dmi_resp_ready = SimDTM_debug_resp_ready; // @[Periphery.scala 166:15:freechips.rocketchip.system.DefaultConfig.fir@207740.4]
  assign ldut_debug_clockeddmi_dmiClock = clock; // @[Periphery.scala 167:20:freechips.rocketchip.system.DefaultConfig.fir@207741.4]
  assign ldut_debug_clockeddmi_dmiReset = reset; // @[Periphery.scala 168:20:freechips.rocketchip.system.DefaultConfig.fir@207742.4]
*/
  assign ldut_debug_clockeddmi_dmi_req_valid = 1'b0;
  assign ldut_debug_clockeddmi_dmi_req_bits_addr = 7'b0;
  assign ldut_debug_clockeddmi_dmi_req_bits_data = 32'b0;
  assign ldut_debug_clockeddmi_dmi_req_bits_op = 2'b0;
  assign ldut_debug_clockeddmi_dmi_resp_ready = 1'b1;
  assign ldut_debug_clockeddmi_dmiClock = 1'b0;
  assign ldut_debug_clockeddmi_dmiReset = 1'b1;

**my testbench:**
tb.v

module tb;
reg reset;

initial	begin
#0	clock=0;
end

always #5	clock=~clock;

initial begin
#0	reset=1;
#200	reset=0;
end

TestHarness dut(
.clock(clock),
.reset(reset),
.io_success(io_success)
);

initial	begin
#10000	$finish;
end

initial	begin
$dumpfile ("tb.vcd");
$dumpvars (0,tb);
end

endmodule

**run simulation:**
%iverilog *.v
%vvp a.out

then VCD waveform generated, use gtkwave to view waveform:
%gtkwave tb.vcd &

If we want to map rocketchip into FPGA, we need to pass this simulation, for SimDTM is not a real synthesizable module.

Can any rocketchip designer help on this, provide a guide how to generate a pure rocketchip rtl, for traditional simulaiton and FPGA implementation? In traditional design flow, we write testbench to wrap the RTL, and load binary program data into SRAM and run simulation.

hcook, 2020-04-09:
`ExampleRocketSystem` is the example top-level synthesizable module class. `TestHarness` is a testbench that instantiates simulation-only constructs like SimDTM and loadable SRAM memories; use it for simulation, but don't try to synthesize it.

Setting `case DebugModuleKey => None` in your `Config` class should remove the Debug Module from inside `ExampleRocketSystem` if that is really what you want.

zhh124, 2020-04-09:
Thanks for hcook's reply. I had search github and found **gsomlo's** project, which had did what I want to do in his/her project:
**https://github.com/gsomlo/yoloRISC**
This project generate the rocketchip RTL with TestHarness module, and hack the external behav srams' size, load user program hex file into sram, and run simulation with verilator, finally I got correct result whit the simulation environment from this project. But when I do simulation with Icarus iverilog, still can't get correct result, I guess it's iverlog simulation's issue or bug.

btw, **gsomlo** also provide a simple_uart module, and use it replace the mmio behav sram, as the output channel for 'hello world' : )

closed 2020-04-09 by zhh124