chipsalliance/rocket-chip #2415: Document command line interface to Scala code

richardxia, 2020-04-14:
**Type of issue**: feature request 

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
Previously, all I had to do was open up `src/main/scala/util/GeneratorUtils.scala` and look for the `App` to find the [command-line arguments](https://github.com/chipsalliance/rocket-chip/blob/93313d910a12d3a8c6e78f657ff305c36e76f2b6/src/main/scala/util/GeneratorUtils.scala#L81-L84).

In the current code, in order to understand the command line arguments without compiling the code myself, I had to do the following:

1. Open up [`GeneratorUtils.scala`](https://github.com/chipsalliance/rocket-chip/blob/301854dcc1be47a6ccc5b98bc177bd05453dc88a/src/main/scala/util/GeneratorUtils.scala). Find that the arguments are no longer there.
2. Try looking at the top-level [`Makefrag`](https://github.com/chipsalliance/rocket-chip/blob/301854dcc1be47a6ccc5b98bc177bd05453dc88a/Makefrag). See that [`PROJECT` is set to `freechips.rocketchip.system`](https://github.com/chipsalliance/rocket-chip/blob/301854dcc1be47a6ccc5b98bc177bd05453dc88a/Makefrag#L7), and take a guess that the entry point might be in that package.
3. Look at [`src/main/scala/system`](https://github.com/chipsalliance/rocket-chip/tree/301854dcc1be47a6ccc5b98bc177bd05453dc88a/src/main/scala/system). See that there is a file named `RocketChipStageGenerator.scala`, which looks promising.
4. Open up [`RocketChipStageGenerator`](https://github.com/chipsalliance/rocket-chip/blob/301854dcc1be47a6ccc5b98bc177bd05453dc88a/src/main/scala/system/RocketChipStageGenerator.scala). See nothing that resembles command-line argument parsing. See a class named `RocketChipCli`, and take a guess that it might have what I'm looking for.
5. Open up [`RocketChipCli`](https://github.com/chipsalliance/rocket-chip/blob/d76703884d9dbd2965536a28a8499928076b89d5/src/main/scala/stage/RocketChipCli.scala). See that the only three things of relevance are `TopModuleAnnotation`, `ConfigsAnnotation`, and `OutputBaseNameAnnotation`.
6. Open up [`RocketChipAnnotations`](https://github.com/chipsalliance/rocket-chip/blob/d76703884d9dbd2965536a28a8499928076b89d5/src/main/scala/stage/RocketChipAnnotations.scala) and look for `TopModuleAnnotation`. Find something that looks like a command line option in [`longOption = "top-module"`](https://github.com/chipsalliance/rocket-chip/blob/d76703884d9dbd2965536a28a8499928076b89d5/src/main/scala/stage/RocketChipAnnotations.scala#L18).
7. Wonder for a moment whether this is GNU-style double hyphen for long options or a Java-style single hyphen for long options.
8. Decide to look for the `ShellOption` class. Curse GitHub's poor search rankings. Eventually find the file that defines it. [Confirm that `longOption`s use double dashes](https://github.com/freechipsproject/firrtl/blob/a7cf6ff3416a11088d811a435ba71fd36b191fb4/src/main/scala/firrtl/options/Registration.scala#L11).

**What is the expected behavior?**
I would like to have a document that describes the command line interface so that I don't have to trace through the code across multiple files and repositories to understand how to run a command. The current README seems to try to avoid describing the actual command-line interface to Rocket and only shows how to run the Makefiles.

**Please tell us about your environment:**
Chrome browser, as I am trying to avoid compiling the code in order to understand its interface.

**What is the use case for changing the behavior?**

mwachs5, 2020-04-14:
Question... understand you don’t want to compile the code. But why would
that have been helpful to you if you had?

richardxia, 2020-04-14:
> Question... understand you don’t want to compile the code. But why would
that have been helpful to you if you had?

Since then I could run `--help` to see the command line arguments:

```
$ sbt 'runMain freechips.rocketchip.system.Generator --help'
[info] Loading settings for project rocket-chip-build from plugins.sbt ...
[info] Loading project definition from /home/rxia/rocket-chip/project
[info] Loading settings for project rocketchip from build.sbt ...
[info] Loading settings for project hardfloat from build.sbt ...
[info] Loading settings for project api-config-chipsalliance from build.sbt ...
[info] Loading settings for project chisel from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/rxia/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] running freechips.rocketchip.system.Generator --help
Usage: rocket-chip [options] [<arg>...]

Shell Options
  <arg>...                 optional unbounded args
  -td, --target-dir <directory>
                           Work directory (default: '.')
  -faf, --annotation-file <file>
                           An input annotation file
  -foaf, --output-annotation-file <file>
                           An output annotation file
  --no-dedup               Do NOT dedup modules
  --show-registrations     print discovered registered libraries and transforms
  --help                   prints this usage text
Logging Options
  -ll, --log-level {error|warn|info|debug|trace}
                           Set global logging verbosity (default: Warn
  -cll, --class-log-level <FullClassName:{error|warn|info|debug|trace}>...
                           Set per-class logging verbosity
  --log-file <file>        Log to a file instead of STDOUT
  -lcn, --log-class-names  Show class names and log level in logging output
Rocket Chip Compiler Options
  -T, --top-module <value>
                           <top module>
  -C, --configs <value>    <comma-delimited configs>
  -n, --name <value>       <base name of output files>
Chisel Front End Options
  -chnrf, --no-run-firrtl  Do not run the FIRRTL compiler (generate FIRRTL IR from Chisel and exit)
  --full-stacktrace        Show full stack trace when an exception is thrown
  --module <package>.<module>
                           The name of a Chisel module to elaborate (module must be in the classpath)
FIRRTL Compiler Options
  -i, --input-file <file>  An input FIRRTL file
  -o, --output-file <file>
                           The output FIRRTL file
  --info-mode <ignore|use|gen|append>
                           Source file info handling mode (default: use)
  -<string>, --firrtl-source <value>
                           An input FIRRTL circuit string
  -X, --compiler <none|high|middle|low|verilog|mverilog|sverilog>
                           The FIRRTL compiler to use (default: verilog)
  -fct, --custom-transforms <package>.<class>
                           Run these transforms during compilation
  -E, --emit-circuit <chirrtl|high|middle|low|verilog|mverilog|sverilog>
                           Run the specified circuit emitter (all modules in one file)
  -e, --emit-modules <chirrtl|high|middle|low|verilog|mverilog|sverilog>
                           Run the specified module emitter (one file per module)
FIRRTL Transform Options
  --no-dce                 Disable dead code elimination
  --no-check-comb-loops    Disable combinational loop checking
  -fil, --inline <circuit>[.<module>[.<instance>]][,...]
                           Inline selected modules
  -clks, --list-clocks -c:<circuit>:-m:<module>:-o:<filename>
                           List which signal drives each clock of every descendent of specified modules
MemLib Options
  -firw, --infer-rw        Enable read/write port inference for memories
  -frsq, --repl-seq-mem -c:<circuit>:-i:<file>:-o:<file>
                           Blackbox and emit a configuration file for each sequential memory

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[success] Total time: 3 s, completed Apr 14, 2020 7:37:18 AM
```

seldridge, 2020-04-14:
This would be an awesome use of `mdoc` to show the output of `Generator --help` on the website. 

More verbose Scaladoc on `RocketChipAnnotation` would also help, like in [other places](https://github.com/freechipsproject/firrtl/blob/master/src/main/scala/firrtl/stage/FirrtlAnnotations.scala#L30).

From the way that the current `Generator`/`RocketChipStage` is constructed, `RocketChipStage extends ChiselStage` (but changes the set of phases that are run). This means that there are command line options for Rocket Chip that come from a lot of different places:

  - `-td`/`--target-dir` which generates a `firrtl.options.TargetDirAnnotation` common to all `Stage`s
  - `-ll`/`--log-level` which generates a `logger.LogLevelAnnotation`
  - `--full-stacktrace` which generates a `chisel3.stage.PrintFullStackTraceAnnotation`

The `firrtl.options` package also exposes a way for FIRRTL transform/library writers to "register" annotations by editing the manifest (META-INF) to make their options discoverable. (Analogous to LLVM passes being able to add options to the LLVM command line.) So, the options available are a function of the class path/manifest...

There's a lot of (I think warranted) architectural complexity here, but a good `man` page auto-generated on the future website would do the trick.

One other cool thing: once Rocket Chip publishing happens, you can theoretically check the command line options for a published version via coursier. Something like:

```bash
coursier \
  launch \
  edu.berkeley.cs:rocketchip_2.12:1.x.y \
  -M freechips.rocketchip.system.Generator \
  -- --help
```