chipsalliance/rocket-chip #2418: dcache's flush all command will do nothing if the 'flushed' signal is asserted?

zhouqinghua, 2020-04-16:
**Type of issue**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
as such instruction sequence:
memory modify -> fence.i -> memory modify again and all hit in dcache -> fence.i
**What is the current behavior?**
the 1st fence.i will flush all dirty cache lines in dcache and do write back. 'flushed' signal will be assereted and keep untill the next tilelink refill request.
the 2nd fence.i will do nothing for the asserting of 'flushed' signal.
so the 2nd round's memory modify dirty cacheline will still stay in dcache. 
**What is the expected behavior?**
the 2nd fence.i should flush all dirty lines. 'flushed' signal should not valid at this time.
'flushed' signal should be cleared when any cacheline is modified instead of watching tilelink's refill request fire.

related codes in DCache.scala:
  // Only flush D$ on FENCE.I if some cached executable regions are untracked.
  if (supports_flush) {
    when (s2_valid_masked && s2_cmd_flush_all) {
      when (!flushed && !io.cpu.s2_kill && !release_ack_wait && !uncachedInFlight.asUInt.orR) {
        flushing := true
        flushing_req := s2_req
      }
    }

  when (tl_out_a.fire() && !s2_uncached) { flushed := false }

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2020-04-17:
There shouldn't be any dirty lines in the D$ is flush is asserted.  Your comment "flushed signal will be assereted and keep untill the next tilelink refill request" is true, but in order to make another line in the cache dirty, you'll have to perform a tilelink refill.

zhouqinghua, 2020-04-28:
if the D$ line  is hit and clean, it needn't perform a tilelink refill. because flush_all will only flush the dirty lines. modify a hit and clean line will not affect the "flushed" signal

zhouqinghua, 2020-04-28:
> There shouldn't be any dirty lines in the D$ is flush is asserted. Your comment "flushed signal will be assereted and keep untill the next tilelink refill request" is true, but in order to make another line in the cache dirty, you'll have to perform a tilelink refill.

if the D$ line is hit and clean, it needn't perform a tilelink refill. because flush_all will only flush the dirty lines. modify a hit and clean line will not affect the "flushed" signal

aswaterman, 2020-04-28:
The cache-flush state machine doesn't just clean the cache; it also invalidates it.  So it's impossible to do another store without a TL transaction.

It is not an optimal implementation of FENCE.I, but it is not really worth improving, because it only gets used when there's no coherence.  (FENCE.I doesn't need to use the state machine when there's coherence.)  Systems without coherence don't tend to spend much time running self-modifying code.

zhouqinghua, 2020-04-29:
> The cache-flush state machine doesn't just clean the cache; it also invalidates it. So it's impossible to do another store without a TL transaction.
>
> It is not an optimal implementation of FENCE.I, [...]

Does that mean the clean lines will be invalidated too? if it does, there will be no problem. i thought it only clean and invalidate the dirty lines.

aswaterman, 2020-04-29:
That’s right. (I also tested this empirically just to make sure.). I’ll close the issue, but feel free to reopen if you have other concerns.

closed 2020-04-29 by aswaterman