chipsalliance/rocket-chip #2425: How to separate the core clock from clock?

yzt000000, 2020-04-21:
Is there an easy way to separate the core clock and bus(axi) clock, just like sifive's U54 .

hcook, 2020-04-21:
If your `Config` changes the value of `RocketCrossingKey`'s [`crossingType` parameter ](https://github.com/chipsalliance/rocket-chip/blob/4c6977a766f96de28e023c6c6837b35259b4aa87/src/main/scala/subsystem/RocketSubsystem.scala#L20) you will get tilelink clock crossings that make it safe to clock the core and uncore differently.

Actually wiring `subsystem.tiles.map(_.clock)` to some clock source other than the implicit Chisel `clock` (e.g. an IO driven externally or a source you have added inside your custom subsystem) is up to you, there is no general purpose solution for supplying clock sources in rocket-chip.

yzt000000, 2020-04-22:
> Actually wiring `subsystem.tiles.map(_.clock)` to some clock source other than the implicit Chisel `clock` (e.g. an IO driven externally or a source you have added inside your custom subsystem) is up to you, there is no general purpose solution for supplying clock sources in rocket-chip.

thanks for your help
So i need add a clock source in  my subsystem, and connect it to tilelink?

hcook, 2020-04-22:
>  and connect it to tilelink?

It depends what you mean by "tilelink". Inside the subsystem there are many tilelink-speaking components: "tiles" that contain cores and caches that speak TileLink at their ports, "buses" that contain crossbars that transmit TileLink operations over an interconnect, "devices" that contain MMIO control register maps and some peripheral functionality. By default, all these things have their clocks driven by the subsystem's  one`clock` pin, which is created implicitly by Chisel.

If you follow my suggestion above, you will get `RocketTile` instances whose tilelink ports can safely transmit messages between clock domains. But now you want to have two or more clock domains inside one subsystem, one for each tile, and then one for everything else. Something's clock source drive must be changed. I suggest only changing the tile clock driver, leaving everything else driven by the implicit `subsystem.clock`, otherwise you will have a lot of work to change all those different components' clocks!

You can change the tiles' clock drivers by adding something like
`outer.tiles.foreach { _.module.clock := my_clock_source_unit.clock }` inside your subclass of `RocketSubsystemModuleImp`. In fact you can even see similar code inside `RocketSubsystemModuleImp` that is driving the tiles' `hartid` and `reset_vector` pins.

yzt000000, 2020-04-23:
> > and connect it to tilelink?
> 
> It depends what you mean by "tilelink". Inside the subsystem there are many tilelink-speaking components: "tiles" that contain cores and caches that speak TileLink at their ports, "buses" that contain crossbars that transmit TileLink operations over an interconnect, "devices" that contain MMIO control register maps and some peripheral functionality. By default, all these things have their clocks driven by the subsystem's one`clock` pin, which is created implicitly by Chisel.
> 
> If you follow my suggestion above, you will get `RocketTile` instances whose tilelink ports can safely transmit messages between clock domains. But now you want to have two or more clock domains inside one subsystem, one for each tile, and then one for everything else. Something's clock pin must be changed be driven from a new source that you define. I suggest only changing the tile clock driver, leaving everything else driven by the implicit `subsystem.clock`, otherwise you will have a lot of work to change all those different components' clocks!
> 
> You can change the tiles' clock drivers by adding something like
> `outer.tiles.foreach { _.module.clock := my_clock_source_unit.clock_out }` inside your subclass of `RocketSubsystemModuleImp`. In fact you can even see similar code inside `RocketSubsystemModuleImp` that is driving the tiles' `hartid` and `reset_vector` pins.

thanks you , got it

closed 2020-05-02 by hcook