chipsalliance/rocket-chip #2467: RocketChipStage.emitVerilog Doesn't Run FIRRTL

seldridge, 2020-05-14:
**Type of issue**: bug report

**Impact**: API Modification

**Development Phase**: request

**Other information**
 

Reported via Gitter ([link](https://view.matrix.org/room/!jSbgpvvpsPbLpsLZOC:matrix.org/?anchor=$tBV3UHZTrE9-MEprtYBUYTIqTRxBWmBrBAR2HRQdruo&highlight)) in reference to a usage of `emitVerilog` ([link](https://github.com/Neurodyne/fusionchip/blob/ebd5ca0340b243585de4c30350c3af8257e93e22/src/main/scala/Emit.scala#L38)).

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

The following should just generate Chirrtl when it is expected that it would generate Verilog:

```scala
class Foo extends RawModule

(new RocketChipStage).emitVerilog(new Foo)
```

**What is the current behavior?**

Using certain methods of `RocketChipStage` inherited from `ChiselStage` won't currently work as intended. 

Consider `emitFirrtl` (which emits High FIRRTL IR) and `emitVerilog` (which emits Verilog for a given design). Because these both rely on running `MaybeFirrtlStage`, but that is excluded via overriding `targets` in `RocketChipStage`, these methods don't work if you use them (nor do they really make sense to use here).

Fundamentally, `ChiselStage.targets` (or the actual `ChiselStage` class) should probably be `final` to avoid this problem. `RocketChipStage` may consequently not want to extend `ChiselStage`.

**What is the expected behavior?**

These methods should either not exist in `RocketChipStage` or they should work as expected.

**Please tell us about your environment:**

`master` branch at time of filing this issue (17c09cc948773b00a054190e428ab5800457893f).

**What is the use case for changing the behavior?**

Prevent user confusion.

debs-sifive, 2020-05-15:
A possible solution is just to make `RocketChipStage` not inherit from `ChiselStage`. The reason I had it inherit was solely to use the `run` function.

I can basically copy/paste Chisel's `run` into `RocketChipStage`, but I'd have to leave out `ErrorLog.errTag` in error messages, as `ErrorLog` is private:
https://github.com/freechipsproject/chisel3/blob/d65013c43f0332e04dd47ce068837b52fbe14cc8/src/main/scala/chisel3/stage/ChiselStage.scala#L48

Not sure how significant that is, though, compared to the fact that it is very confusing that `RocketStage` can't use the functions you listed. Thoughts on this approach, @seldridge?

jackkoenig, 2020-05-15:
Generally composition is better than inheritance, `RocketChipStage.run` could simply call `ChiselStage.run`.

seldridge, 2020-05-15:
I think the `run` method is only useful for elaboration, yes? If that's so, then I think the following would work:

1. Define a new private class that extends `ChiselStage` and overrides the existing targets just adding `GenerateROMs`
2. Change `RocketChipStage` to not inherit from `ChiselStage` and instead use the private `ChiselStage` child internally
3. Add a new `AddDeafults` phase that adds a `NoRunFirrtlCompilerAnnotation` annotation 
4. (**Future Work**) Extend the `RocketChipStage` CLI with an option that let's you disable the `NoRunFirrtlCompilerAnnotation` insertion so that a user could choose to go right to Verilog. This can then add helper methods back in like `emitChirrtl`, `emitFirrtl`, and `emitVerilog` but taking a Rocket Chip-appropriate argument list (a module/lazy module and a configuration).

seldridge, 2020-05-15:
@jackkoenig: I think that's the ideal approach. The current problem is that `ChiselStage`'s emitter will delete the `ChiselCircuitAnnotation` which `GenerateROMs` needs to work. There are three workarounds here:

1) Re-run elaboration :(
2) Examine the deleted annotation (janky)
3) Custom ChiselStage (approach I was suggesting above)

The [how-to-serialize PR](https://github.com/freechipsproject/firrtl/pull/1277) is intended to fix this by removing the emitter phase.

debs-sifive, 2020-05-18:
@seldridge I tried @jackkoenig's composition suggestion, and it seems to work.

> The current problem is that ChiselStage's emitter will delete the ChiselCircuitAnnotation which GenerateROMs needs to work.

I think dependency issues are orthogonal to the `run` function, since all the function seems to do is call the phase manager. Also, it looks like `Convert` is the stage that deletes `ChiselCircuitAnnotation`, and `Convert` is currently listed as a dependent of `GenerateROMs`.

PR #2481 (2020-05-28): RocketChipStage shouldn't extend ChiselStage
Add a separate `RocketChiselStage` that adds appropriate modifications to `ChiselStage` that rocket chip needs (only modification is the addition of . Remove `ChiselStage` from `RocketChipStage`'s type hierarchy.

I only spot checked that this passes `DefaultConfig` and that the stack trace trimming didn't appear to break.

I think that this is a better approach than to rely on access to `ChiselStage.run` which is a method that should probably be made `protected`.

**Related issue**: Fixes #2467

**Type of change**: other enhancement

**Impact**: AP [... cut]
closed 2020-05-28 by debs-sifive