chipsalliance/rocket-chip #2514: Error when running

tswanson4444, 2020-06-11:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request 

**Other information**
 
[info] [0.002] Elaborating design...
[error] java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined in Parameters
[error] hisel / ...pile / unmanagedSources / allInputPathsAndAttributes 0s
[error] 	at chipsalliance.rocketchip.config$View.apply(Config.scala:28)
[error] 	at chipsalliance.rocketchip.config$View.apply(Config.scala:25)
[error] 	at freechips.rocketchip.subsystem.HasConfigurableTLNetworkTopology.$init$(BaseSubsystem.scala:64)
[error] 	at freechips.rocketchip.subsystem.BaseSubsystem.<init>(BaseSubsystem.scala:75)
[error] 	at freechips.rocketchip.subsystem.RocketSubsystem.<init>(RocketSubsystem.scala:78)
[error] 	at freechips.rocketchip.system.ExampleRocketSystem.<init>(ExampleRocketSystem.scala:14)
[error] 	at freechips.rocketchip.system.TestHarness.<init>(TestHarness.scala:16)
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error] 	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error] 	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error] 	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:31)
[error] 	... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
make verilog CONFIG=freechips.rocketchip.system.DefaultFPGAConfig
**What is the current behavior?**
During elaborating design the make with crash.
**What is the expected behavior?**
Generate Verilog files

CLappin, 2020-06-11:
Hi 

try:
make verilog CONFIG=DefaultFPGAConfig

tswanson4444, 2020-06-11:
> Hi
> 
> try:
> make verilog CONFIG=DefaultFPGAConfig
This does not fix the issue.  On a side note freechips.rocketchip.system.DefaultConfig runs without issues

yang-le, 2020-06-14:
Hi @tswanson4444 
you can try this

diff --git a/src/main/scala/system/Configs.scala b/src/main/scala/system/Configs.scala
index b76507bb0..c0bbc13c8 100644
--- a/src/main/scala/system/Configs.scala
+++ b/src/main/scala/system/Configs.scala
@@ -74,5 +74,5 @@ class MMIOPortOnlyConfig extends Config(
   new DefaultConfig
 )

-class BaseFPGAConfig extends Config(new BaseConfig)
+class BaseFPGAConfig extends Config(new BaseConfig ++ new WithCoherentBusTopology)
 class DefaultFPGAConfig extends Config(new WithNSmallCores(1) ++ new BaseFPGAConfig)

It seems that they forget to add WithCoherentBusTopology into BaseFPGAConfig.

tswanson4444, 2020-06-15:
Thanks!  This fixed the issue.

closed 2020-06-15 by tswanson4444