chipsalliance/rocket-chip #2580: Rocketchip TLB not dealing with PTE A and D bit?

njuallen, 2020-07-29:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: request

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

Hi, I have been digging rocketchip TLB implementation for a while. I'm quite confused by the way rocketchip dealing with PTE A and D bit.
Accoring to the spec(https://riscv.org/specifications/privileged-isa/), page 69, section 4.3.1 on the update of PTE A and D bit. 
The processor can either throw a exception and let OS update PTE A and D bit. Or it can update it with very complex hardware.
I read rocketchip's TLB implementation, but It seems that: Rocketchip's TLB does not read, check or update PTE A and D bit. So I think it does not conform to the spec. But for some reason, linux still runs well. 

Can anyone tell me:
Is rocketchip's TLB implementation spec compliant?
Why linux still runs?

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

aswaterman, 2020-07-29:
It’s conformant. The PTW handles it, not the TLB.

njuallen, 2020-07-30:
Thanks for you answer, I will go check it.

closed 2020-08-07 by njuallen

MaxBanister, 2022-05-31:
Is there any documentation on where this occurs? I am looking for where the exception is generated/bits are set in the PTW code