chipsalliance/rocket-chip #2599: DefaultFPGAConfig part not found

Sinbiosis, 2020-08-10:
**Type of issue**: bug report

**Impact**: FPGA Building

**Other information**
 
lore@Helium:~/src/rocket-chip/vsim$ make verilog CONFIG=DefaultFPGAConfig
mkdir -p /home/lore/src/rocket-chip/vsim/generated-src/
cd /home/lore/src/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/lore/src/rocket-chip/sbt-launch.jar "runMain freechips.rocketchip.system.Generator -td /home/lore/src/rocket-chip/vsim/generated-src -T freechips.rocketchip.system.TestHarness -C DefaultFPGAConfig"
OpenJDK 64-Bit Server VM warning: Ignoring option MaxPermSize; support was removed in 8.0
[info] Loading settings for project rocket-chip-build from plugins.sbt ...
[info] Loading project definition from /home/lore/src/rocket-chip/project
[info] Loading settings for project rocketchip from build.sbt ...
[info] Loading settings for project hardfloat from build.sbt ...
[info] Loading settings for project api-config-chipsalliance from build.sbt ...
[info] Loading settings for project chisel from build.sbt ...
Using addons: 
[info] Set current project to rocketchip (in build file:/home/lore/src/rocket-chip/)
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
[info] running freechips.rocketchip.system.Generator -td /home/lore/src/rocket-chip/vsim/generated-src -T freechips.rocketchip.system.TestHarness -C DefaultFPGAConfig
[error] (run-main-0) java.lang.Exception: Unable to find part "DefaultFPGAConfig" from "ArrayBuffer(DefaultFPGAConfig)", did you misspell it or specify the wrong package path?
[error] java.lang.Exception: Unable to find part "DefaultFPGAConfig" from "ArrayBuffer(DefaultFPGAConfig)", did you misspell it or specify the wrong package path?
[error] 	at Chisel.package$throwException$.apply(compatibility.scala:436)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.$anonfun$getConfig$1(GeneratorUtils.scala:19)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight(IndexedSeqOptimized.scala:65)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight$(IndexedSeqOptimized.scala:72)
[error] 	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:49)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig(GeneratorUtils.scala:14)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig$(GeneratorUtils.scala:13)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.getConfig(PreElaboration.scala:16)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:26)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:16)
[error] 	at firrtl.options.DependencyManager.$anonfun$transform$3(DependencyManager.scala:267)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft(LinearSeqOptimized.scala:126)
[error] 	at scala.collection.LinearSeqOptimized.foldLeft$(LinearSeqOptimized.scala:122)
[error] 	at scala.collection.immutable.List.foldLeft(List.scala:89)
[error] Caused by: java.lang.ClassNotFoundException: DefaultFPGAConfig
[error] 	at java.base/java.net.URLClassLoader.findClass(URLClassLoader.java:471)
[error] 	at sbt.internal.ManagedClassLoader.findClass(ManagedClassLoader.java:92)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:589)
[error] 	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)
[error] 	at java.base/java.lang.Class.forName0(Native Method)
[error] 	at java.base/java.lang.Class.forName(Class.java:315)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.$anonfun$getConfig$1(GeneratorUtils.scala:16)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight(IndexedSeqOptimized.scala:65)
[error] 	at scala.collection.IndexedSeqOptimized.foldRight$(IndexedSeqOptimized.scala:72)
[error] 	at scala.collection.mutable.ArrayBuffer.foldRight(ArrayBuffer.scala:49)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig(GeneratorUtils.scala:14)
[error] 	at freechips.rocketchip.util.HasRocketChipStageUtils.getConfig$(GeneratorUtils.scala:13)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.getConfig(PreElaboration.scala:16)
[error] 	at freechips.rocketchip.stage.phases.PreElaboration.transform(PreElaboration.scala:26)
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 11 s, completed Aug 10, 2020, 12:52:22 AM

**If the current behavior is a bug, please provide the steps to reproduce the problem:**
cd $ROCKETCHIP/vsim
make verilog CONFIG=DefaultFPGAConfig 
**What is the current behavior?**
An error is thrown because the build can't "find part DefaultFPGAConfig"
**What is the expected behavior?**
That the make command produces a verilog file intended to be used in an FPGA flow
**Please tell us about your environment:**

version: commit f3c75b30ffb3542dd983546b543b1b6d5acf2c29
OS: linux x86_64 5.4.0-42-generic
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.1 LTS
Release:	20.04
Codename:	focal
**What is the use case for changing the behavior?**
Use in FPGA flows like Vivado

Sinbiosis, 2020-08-11:
Okay. I don't know why but I didn't read that you have to use freechips.rocketchip.system.DefaultFPGAConfig not just DefaultFPGAConfig. However, then I get an error 

[info] [0.003] Elaborating design...
[error] java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined in Parameters

What is wrong now?

hcook, 2020-08-13:
Oops, looks like that Config got broken because it isn't covered by this repository's CI, sorry. The fix is simple and I opened #2603 to fix it.

I will also mention that that Config is not sufficient to get a working FPGA imagine on any particular board on its own, so you might want to take a look at the https://github.com/sifive/freedom or https://github.com/firesim/firesim projects (which build on this one) and provide a more complete set of design resources for functionally complete FPGA designs (IO peripherals, DDR controllers, etc).

HaFred, 2021-09-24:
Hi @hcook , sorry to bring this up.
I ran into the same issue as @Sinbiosis did, and I tried the fix in #2603 , but the make error still existed. Any idea on what is going now? I was running with the master branch.