chipsalliance/rocket-chip #2600: type Config is not found when compling emulation

netanelgonen, 2020-08-10:
**Type of issue**: bug report

**Other information**
 

install rocket tools and try to compile the emulation according to instructions 

getting the error:
[info] Compiling 1 Scala source to /.autodirect/mtrswgwork/netanelg/proj/rocket-chip/api-config-chipsalliance/build-rules/sbt/target/scala-2.12/classes ...
[error] /.autodirect/mtrswgwork/netanelg/proj/rocket-chip/api-config-chipsalliance/build-rules/sbt/../../design/craft/src/config/Config.scala:102:32: not found: type Config
[error] class DefaultConfigRBB extends Config(
[error]                                ^
[error] /.autodirect/mtrswgwork/netanelg/proj/rocket-chip/api-config-chipsalliance/build-rules/sbt/../../design/craft/src/config/Config.scala:103:47: no arguments allowed for nullary constructor Object: ()Object
[error] new WithJtagDTMSystem ++ new WithNBigCores(1) ++ new BaseConfig)
[error]                                               ^
[error] /.autodirect/mtrswgwork/netanelg/proj/rocket-chip/api-config-chipsalliance/build-rules/sbt/../../design/craft/src/config/Config.scala:105:33: not found: type Config
[error] class QuadCoreConfigRBB extends Config(
[error]                                 ^
[error] /.autodirect/mtrswgwork/netanelg/proj/rocket-chip/api-config-chipsalliance/build-rules/sbt/../../design/craft/src/config/Config.scala:106:47: no arguments allowed for nullary constructor Object: ()Object
[error] new WithJtagDTMSystem ++ new WithNBigCores(4) ++ new BaseConfig)
[error]                                               ^
[error] four errors found
[warn] Multiple main classes detected.  Run 'show discoveredMainClasses' to see the list
^C
[warn] Canceling execution...
[error] java.util.concurrent.ExecutionException: Incomplete(node=None, tpe=Error, msg=Some(cancelled), causes=List(), directCause=None)
[error] Use 'last' for the full log.

hcook, 2020-08-27:
You need to add the example DefaultConfigRBB
to https://github.com/chipsalliance/rocket-chip/blob/master/src/main/scala/system/Configs.scala
not https://github.com/chipsalliance/api-config-chipsalliance/blob/master/design/craft/src/config/Config.scala

Looks like #2209 made things more confusing and not less

hcook, 2020-08-27:
Closed by #2620

closed 2020-08-27 by hcook