chipsalliance/rocket-chip #2629: Emulator make error

jerryhethatday, 2020-09-04:
**Type of issue**: bug report

**Impact**: no functional change 

**Development Phase**: request
**Other information**
 
I followed the README instructions of the Rocket chip Repo:
1):clone the wholle repo first
git clone https://github.com/ucb-bar/rocket-chip.git
cd rocket-chip
#export ROCKETCHIP='pwd' #I didn't execute this LOC
git submodule update --init  

2) clone the toolchain repo and make it
git clone https://github.com/freechipsproject/rocket-tools
cd rocket-tools
git submodule update --init --recursive
export RISCV=/path/to/install/riscv/toolchain
export MAKEFLAGS="$MAKEFLAGS -j4" # 4cores in the VMware ubuntu
./build.sh
#./build-rv32ima.sh (if you are using RV32)  #I didn't execute this LOC, I have no idea #what it mean by "if you are using RV32". if I am using RV64, then this LOC wont be #needed?

after I type “./build.sh", it takes while and prints lots of warnings, but finally it completed.
3)build the emulator
**cd emulator
make**  
it took 10mins or so, and I thought everything wokred just fun, but it went wrong, the terminal printed stuff like following:
/usr/bin/install: cannot create regular file '/home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/verilator/examples/xml_py/sub.v': No such file or directory
/usr/bin/install: cannot create regular file '/home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/verilator/examples/xml_py/top.v': No such file or directory
/usr/bin/install: cannot create regular file '/home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/verilator/examples/xml_py/Makefile': No such file or directory
/bin/sh ./src/mkinstalldirs /home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/pkgconfig
mkdir /home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/pkgconfig
/usr/bin/install -c -m 644 verilator.pc /home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/pkgconfig
/usr/bin/install -c -m 644 verilator-config.cmake /home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/verilator
/usr/bin/install -c -m 644 verilator-config-version.cmake /home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/share/verilator
make[1]: Leaving directory '/home/jerry/rocketchipGen/rocket-chip/emulator/verilator/src/verilator-4.028'
touch /home/jerry/rocketchipGen/rocket-chip/emulator/verilator/install/bin/verilator
cd /home/jerry/rocketchipGen/rocket-chip/emulator/generated-src && \
/home/jerry/rocketchipGen/rocket-chip/scripts/vlsi_mem_gen /home/jerry/rocketchipGen/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.conf > /home/jerry/rocketchipGen/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v.tmp && \
mv -f /home/jerry/rocketchipGen/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v.tmp /home/jerry/rocketchipGen/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v
**/usr/bin/env: ‘python’: No such file or directory**
Makefrag-verilator:28: recipe for target '/home/jerry/rocketchipGen/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v' failed
make: *** [/home/jerry/rocketchipGen/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfig.behav_srams.v] Error 127

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**
Linux ubuntu 4.18.0-15-generic #16~18.04.1-Ubuntu SMP Thu Feb 7 14:06:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

**What is the use case for changing the behavior?**

sequencer, 2020-10-09:
> /usr/bin/env: ‘python’: No such file or directory

You didn't install python in your system.
try `apt install python` or something else.

deadsupra, 2020-11-14:
Why is python 2 still used?

closed 2021-03-22 by jerryhethatday