chipsalliance/rocket-chip #2637: Rocketchip behavior when reading pmpcfg2 register.

JaewonHur, 2020-09-15:
**Type of issue**: bug report 

**Impact**: unknown

**Development Phase**: request 

**Other information**
 

I am using Rocketchip default configuration from the commit `3f03ba`.

I think it is not the specification issue but 
there is a behavior inconsistency between Spike and Rocket.

When configuring both to use only 8 pmp registers.
Spike raise illegal instruction exception when trying to access `pmpcfg2` but Rocketchip doesn't.

I wonder is there any plan to make them consistent.
Here is the testcase.

[test.zip](https://github.com/chipsalliance/rocket-chip/files/5222587/test.zip)

aswaterman, 2020-09-15:
In the test I wrote, pmpcfg2 was accessible in Spike.

JaewonHur, 2020-09-16:
Ah, thanks.

closed 2020-09-16 by JaewonHur