chipsalliance/rocket-chip #2658: how to stall/restart the pipeline of Rocket?

meton-robean, 2020-10-09:
what I want to do is that: 
when encountering a specific instruction address at Ins-Decode stage,  I want to stop the pipeline and handle something, after finishing it, I restart the pipeline so that the program continues to execute.

how can I achieve the operations above?  Which wires do I need to set when I stop/restart the pipeline?

aswaterman, 2020-10-09:
Look at the `ctrl_stalld` signal... but please note that what you're trying to do isn't a supported feature, so you're on your own.

meton-robean, 2020-10-09:
> Look at the `ctrl_stalld` signal... but please note that what you're trying to do isn't a supported feature, so you're on your own.

do I need to clear the ibuf when I restart the pipeline?

closed 2020-10-17 by meton-robean