chipsalliance/rocket-chip #2670: Generating the Remote Bit-Bang (RBB) Emulator not working properly

mikemusasa, 2020-10-19:
**Type of issue**: bug report 

**Impact**:  unknown

**Development Phase**: request 

**Other information**
I followed the instructions in the read me file and al the steps as metioned, 
I added as suggested   in  src/main/scala/system/Configs.scala 
   class DefaultConfigRBB extends Config(
new WithJtagDTMSystem ++ new WithNBigCores(1) ++ new BaseConfig)

class QuadCoreConfigRBB extends Config(
new WithJtagDTMSystem ++ new WithNBigCores(4) ++ new BaseConfig)

**What is the current behavior?**
when running the command :   CONFIG=freechips.rocketchip.system.DefaultConfigRBB make  ,   I am receiving this error message : 

      [error] java.lang.IllegalArgumentException: requirement failed: Key TLNetworkTopologyLocated(Location(InSubsystem)) is not defined in Parameters
[error] ocket-ma...s / Compile / compileIncremental 0s
[error]         at chipsalliance.rocketchip.config$View.apply(Config.scala:28)
[error]         at chipsalliance.rocketchip.config$View.apply(Config.scala:25)
[error]         at freechips.rocketchip.subsystem.HasConfigurableTLNetworkTopology.$init$(BaseSubsystem.scala:64)
[error]         at freechips.rocketchip.subsystem.BaseSubsystem.<init>(BaseSubsystem.scala:75)
[error]         at freechips.rocketchip.subsystem.RocketSubsystem.<init>(RocketSubsystem.scala:38)
[error]         at freechips.rocketchip.system.ExampleRocketSystem.<init>(ExampleRocketSystem.scala:13)
[error]         at freechips.rocketchip.system.TestHarness.<init>(TestHarness.scala:16)
[error]         at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
[error]         at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
[error]         at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
[error]         at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
[error]         at freechips.rocketchip.stage.phases.PreElaboration.$anonfun$transform$1(PreElaboration.scala:31)
[error]         ... (Stack trace trimmed to user code only, rerun with --full-stacktrace if you wish to see the full stack trace)

Exception: sbt.TrapExitSecurityException thrown from the UncaughtExceptionHandler in thread "run-main-0"
[error] Nonzero exit code: 1
[error] (Compile / runMain) Nonzero exit code: 1
[error] Total time: 16 s, completed Oct 19, 2020, 7:55:02 AM
make: *** [Makefrag-verilator:13: /home/mikemusasa/work/rocket-chip/emulator/generated-src/freechips.rocketchip.system.DefaultConfigRBB.fir] Error 1
   
**What is the expected behavior?**

Suppose to finish the make and provide me the ability to use the DefaultConfigRBB that was just created . 

Please tell us about your environment : 

running everything using  Ubuntu 20.04 LTS 

-->
**What is the use case for changing the behavior?**

sequencer, 2020-11-06:
You didn't assign your `BusTopology`, so key `TLNetworkTopologyLocated` will be missing.
Try to append any of these configs: `WithCoherentBusTopology`, `WithIncoherentBusTopology`, `WithJustOneBus`.

mikemusasa, 2020-11-06:
how do i append that ? is it done  when calling the  same line  CONFIG=freechips.rocketchip.system.DefaultConfigRBB make   ?  this was not mentioned in the instructions on the readme

mikemusasa, 2020-11-06:
disregard my last comment , i made it work now but some corrections are needed to the Read me files in order for others to not have this same issue

sequencer, 2020-11-06:
Thank you for pointing out!

closed 2020-11-18 by sequencer