chipsalliance/rocket-chip #2681: reg_misa not correctly encoded

shahzaibk23, 2020-10-24:
**Type of issue**: I've generated an SoC with my custom configuration as follows , extensions: i,m,c , RV-32 but now when i try to run it , and generate a .vcd file for simulation i'm seeing, the reg_misa base fields(MSBs) as 11 (equivalent to 128-bit)  but in actual it should be 01( for 32-bit).

What I may be doing wrong? What should i do in order to resolve this issue?

aswaterman, 2020-10-24:
The contents of `reg_misa` are a microarchitectural detail.  Did you try executing software that reads the value of `misa` using a `csrr` instruction and observe the same phenomenon?

shahzaibk23, 2020-10-24:
Yes, I viewed the .vcd file in GTKWave, and in the reg_misa base address (MSBs) are 11.
I compared these values to the RISC-V Privileged Manual, but the values aren't correct according to that.
The reg_misa base address (MSBs) should be 01, because my configuration is of 32-bit, but in v.cd 11 is being shown , which is for 128-bit (According to the Manual)

aswaterman, 2020-10-24:
That's not what I'm asking.  Did you try executing a `csrr t0, misa` instruction on the core and verify that bits 31:30 of `t0` aren't what you expect?

closed 2020-10-25 by shahzaibk23