chipsalliance/rocket-chip #2690: How can I reduce Rocket Chip DRAM size from 256MB ?

Uzair540, 2020-10-29:
I am having integer overflow error because of the large 256MB size of DRAM. 
[image]
I am using 64kb data and instruction cache I want to reduce DRAM size from where in scala/Chisel file I can make changes? 
Line 224 gives me an error of integer flow which means my DRAM size is very large 256MB.How can I change my DRAM size from my scala code Can you elaborate me the path  ?
@aswaterman @yunsup

aswaterman, 2020-10-29:
https://github.com/chipsalliance/rocket-chip/blob/630600d860a0ff0ec1e290931d301cc6f2b96fe0/src/main/scala/subsystem/Configs.scala#L392

Uzair540, 2020-10-31:
@aswaterman can you please elaborate what kind of information this variable is giving about the size of RAM,How many address bits are there and width of the memory?

aswaterman, 2020-11-01:
That's the number of bytes in the main memory.  Reducing that number will reduce the number of rows in the RAM.

Uzair540, 2020-11-04:
thanks I got your point

closed 2020-11-04 by Uzair540