chipsalliance/rocket-chip #2691: Behavior mismatch between Rocket Chip and spike

JaewonHur, 2020-10-29:
**Type of issue**: bug report

**Impact**: unknown

**Development Phase**: proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

During running fuzzing on Rocket Chip. 
I found a curious behavior.

In the test case I give you, medeleg register is set to 0xb100 (1011 0001 0000 0000) 
which does not contain 'Store/AMO address misaligned'.
But when I run the executable on RISC-V Rocket CPU,
The CPU hang because It transfers the store address misaligned trap to stvec.
(Boom and spike behave differently.)

In addition, in my benchmark, the case is reproduced only when using seed 0.
Can you check this behavior?

You can compile test case as follow in the given directory.
(I was able to reproduce it with seed 0, but in some case with random seed it is not reproduced.)
`make test_v.riscv `
`./simulator-chipyard-RocketConfig --seed=0 +verbose test_v.riscv`

I'm using chipyard repo commit `2c0928c`.

[test.zip](https://github.com/chipsalliance/rocket-chip/files/5457135/test.zip)

aswaterman, 2020-10-29:
It seems the reason the test passes with some seeds and not others is that the FCSR is not initialized, so the rounding mode might be invalid.

The address you are accessing causes a page fault.  Rocket gives higher priority to page faults than misaligned exceptions, for the following reason.  The ISA spec allows misaligned accesses to raise either misaligned exceptions or access exceptions, depending on the physical address.  (This allows you to prevent misaligned accesses to I/O devices from being emulated in software.)  But to know the physical address, you must first translate the address, which in this case causes a page fault.

Both Rocket and Spike are correct.

JaewonHur, 2020-10-30:
Ah, thanks.

closed 2020-10-30 by JaewonHur