chipsalliance/rocket-chip #2695: IO Placement failed due to overutilization

Uzair540, 2020-11-01:
I have got an error when making ExampleRocketSystem as top.
[image]
The error is as follow
[image]
Can you suggest me how can I resolve this error?
@aswaterman  @yunsup

aswaterman, 2020-11-01:
We can't offer support for using rocket-chip on FPGA, but the basic issue is that it doesn't make sense to map rocket-chip to an FPGA without writing additional Verilog that connects the various top-level ports from ExampleRocketSystem to the various resources on the FPGA SoC.  For an example, you could check out https://github.com/sifive/freedom, but note it uses an older version of rocket-chip.  But again, we don't have resources to support you in this effort.

Uzair540, 2020-11-01:
If I am not wrong the **TestHarness** module provides us the peripherals in an SoC which are connected with the Core (verilog modules you are refering above).

aswaterman, 2020-11-01:
Yes, but they're simulation constructs, not meant to be synthesized.

Uzair540, 2020-11-01:
ok thanks

closed 2020-11-01 by Uzair540