chipsalliance/rocket-chip #2703: Core connection with DCCM, ICCM AND GPIO

Uzair540, 2020-11-04:
@aswaterman I want to make simple SOC having DCCM,ICCM and GPIOs can you elaborate me how can I make changes in my config file to add these modules

sequencer, 2020-11-04:
DCCM and ICCM can be added by directly attach SRAM under cores. you can attach TLRAM to the correct place.
GPIO is implemented in https://github.com/sifive/sifive-blocks, which you can take a look.
Since Rocket-Chip has a lot of possibility to explore, I think upstream don't have enough resources on each possibility of design.

aswaterman, 2020-11-04:
You should ask @yunsup about that

sequencer, 2020-11-09:
@Uzair540 
FYI: 
DCCM, which in RocketChip should be named DTIM. related config:
https://github.com/chipsalliance/rocket-chip/blob/58a8d8f98b854a794c4ccf0cbde8af3f76043fc5/src/main/scala/rocket/HellaCache.scala#L39
ICCM, which in RocketChip should be named ITIM. related config:
https://github.com/chipsalliance/rocket-chip/blob/58a8d8f98b854a794c4ccf0cbde8af3f76043fc5/src/main/scala/rocket/ICache.scala#L30

But configuration to these two parameter is not supported by RocketChip out of box.

Uzair540, 2020-11-15:
Thanks for the help :+1:

closed 2020-11-15 by Uzair540