chipsalliance/rocket-chip #2704: How can I measure the miss penalties for microarchitectural events?

yunchen-juuuump, 2020-11-05:
Hi

I would like to configure the Cache/DRAM access latencies manually, but I don't know where can I configure them. I find this [documentation](https://inst.eecs.berkeley.edu/~cs152/sp18/handouts/lab2-0.2.pdf) that shows the memory access latency(in Table.3). So, where can I configure these latencies? 

Thanks

sequencer, 2020-11-06:
I think this question is not about rocket-chip.
Your cache access latency is already decided in the cache micro architectural, which is not configurable, since rocket-chip is the RTL generator.
Your DRAM access latency is defined in your simulation environment, as for rocket-chip, it is based on the `SimAxiMem`, you can add additional Buffer as the client of your memory. Or just add delays to the `AXI4RAM`.

yunchen-juuuump, 2020-11-06:
@sequencer, thanks for your reply. So, if I would like to configure the access latency of AIX4RAM, what should I do? I find a variable named **minLatency** in AIX4RAM, is this the access latency? Should I modify it? Or should I use **Delayer** in axi4 to enable the access delay? Thanks!

sequencer, 2020-11-09:
Sorry for the late reply I missed this issue.
My personal suggestion is that you can add a `TLDelayer` under your coherence manger which should be directly connected to memory port.
`minLatency` is just for computation, it doesn't affect hardware construction.

sequencer, 2020-11-19:
Feel free to reopen this issue if have more questions about this.

closed 2020-11-19 by sequencer