chipsalliance/rocket-chip #2705: mulUnroll variable not understand

Uzair540, 2020-11-05:
mulUnroll: Int = 1
  mulEarlyOut: Boolean = false,
  divEarlyOut: Boolean = false
I have some issue in understanding these mulUnroll, mulEarlyOut and divEarlyOut in multiplier module.Can you explain me what it is exactly doing in the code ? 
@aswaterman @hcook @yunsup

aswaterman, 2020-11-06:
The number of cycles per multiplication is roughly (xLen / mulUnroll).  So if you set mulUnroll = 64 for an RV64 core, you'll get a fully pipelined multiplier.

If the multiplier isn't fully pipelined (i.e., mulUnroll < xLen), then mulEarlyOut will make the multiplier's latency data-dependent, i.e., it'll take fewer cycles to multiply small numbers than large ones.  If the multiplier is fully pipelined, this option is irrelevant.

divEarlyOut is analogous to mulEarlyOut (smaller numbers divide more quickly than larger ones).

Since I think you're targeting an FPGA that has hardened multiplier arrays, my recommendation is to use mulUnroll = 64, mulEarlyOut = false (but it's a don't-care), and divEarlyOut = whatever you want (setting it to true affects the Dhrystone score by 10%).

Uzair540, 2020-11-08:
Thanks.I got it..

closed 2020-11-08 by aswaterman

Uzair540, 2021-04-15:
@hadirkhan10