chipsalliance/rocket-chip #2707: rocket can not run

whutmsy, 2020-11-06:
i want make rocket into fpga accelerator. i use rocket-chip, testchipip and zynqadapter. when i simulate on vivado, the signal of rocket core is X. it has clock and reset. Why my rocket can not running.

sequencer, 2020-11-06:
Would you mind providing more information? From currently information, we cannot debug with only one line problem description.

whutmsy, 2020-11-06:
@sequencer thanks ! I package the rocket as a vitis rtl kernel. so i make the rocket only has a axi4 port. this port is the serial IO from testchipip. and i use the zynqadapter change the serial io to axi4 port. This is almost from fpga-zynq(https://github.com/ucb-bar/fpga-zynq). so my top module is only one axi4 port including clock and reset.

this is the tile waveform
[image]
this is the core waveform. 
[image]

sequencer, 2020-11-06:
We mostly use diplomacy as uncore SoC generator, you drive your AXI with Xilinx part. You may need to take a look at your adapter part. And Vivado simulator or modelsim is also unsupported. So I think this is not a bug in rocket-chip.

My suggestion is take a look at fpga-shells rather than rocket-zynq. And use diplomacy to construct your desired SoC, rather than Xilinx bus IP.

whutmsy, 2020-11-06:
@sequencer  thanks for your suggestion. The AXI4 port is used to write program into 0x80000000, and interrupt the rocket to run. i checked the data is rigth. when the rocket start to boot, I wonder if the rocket-chip run only need a clock, or any other signals.

sequencer, 2020-11-06:
You also need reset.

whutmsy, 2020-11-06:
@sequencer   thank you!

sequencer, 2020-11-06:
Is this problem resolved?

whutmsy, 2020-11-06:
@sequencer not yet. I am checking my waveform.  I have reset and clock signal. And the reset is from 1 to 0.  why the rocket can not boot. i give the bootrom, and 0x10040 is the beginning of bootrom. But the signal is X.

whutmsy, 2020-11-06:
this is the tile .
[image]

sequencer, 2020-11-07:
Since there is no enough information about your environments and Xilinx Bus IP, I'm not able to help you on this.
You can take a look and try to understand verilator based simulation codes in
https://github.com/chipsalliance/rocket-chip/blob/master/vsim/Makefrag
And port to another simulator and bus interfaces.

whutmsy, 2020-11-07:
@sequencer  I migth ignore some module. I use the chipyard verilator simulator using rocketconfig (https://github.com/ucb-bar/chipyard/blob/master/generators/chipyard/src/main/scala/config/RocketConfigs.scala), and i got right result.  I take out the processor and give clock and reset signal, finally got the same result as my previous simulation.

This is the top module.
[image]

whutmsy, 2020-11-07:
this is my Top module
[image]

sequencer, 2020-11-08:
Sorry, I think I’m unable to help you on this, since the problem of you question is out of the rocket-chip.
My suggestion is you need to understand the simulation interface(by understanding verilator simulation(supported by rocket-chip) firstly) and adapter of Xilinx bus IP, before your own try on other simulation tool and IPs.
If you have any other question on codes related to rocket-chip, we may help you on this.

closed 2020-11-08 by sequencer