chipsalliance/rocket-chip #2708: How can I make verilog into separate modules rather than into one

cc910228, 2020-11-06:
I am used "make verilog" command to convert rocket chip into verilog.
for now it gives a entire verilog file named "freechips.rocketchip.system.DefaultConfig.v"
but I prefer to separate them into individual modules. 
does anyone  can give some sugguestions?

sequencer, 2020-11-06:
rocketchip doesn't natively support this, but you can try to add "--emit-modules" to FIRRTL command line.

cc910228, 2020-11-06:
> rocketchip doesn't natively support this, but you can try to add "--emit-modules" to FIRRTL command line.

I am sorry but I am a beginner for this, could you please tell me more about the detail?
I tried to search for it and I get to know that I can us -fsm (which is "--emit-modules" you mentioned) for the FIRRTL command line. 
But I am not sure how can I find the FIRRTL command line.
I am using "make verilog" in the "vsim" folder and it automatically generated the verilog files. In this folder there are "makefile", "makefrag", and "makefrag-verilog". I looked at all of them and I couldn't find where to add the command.

Looking forward to your reply

sequencer, 2020-11-07:
Here is a work around diff for `vsim/makefrag-verilog`, but this usage is unsupported.
```diff
--- a/vsim/Makefrag-verilog
+++ b/vsim/Makefrag-verilog
@@ -15,8 +15,7 @@ $(generated_dir)/%.fir $(generated_dir)/%.d: $(ROCKET_CHIP_JAR) $(bootrom_img)
 $(generated_dir)/%.v $(generated_dir)/%.conf: $(generated_dir)/%.fir $(ROCKET_CHIP_JAR)
        mkdir -p $(dir $@)
        $(FIRRTL) -i $< \
-    -o $(generated_dir)/$*.v \
-    -X verilog \
+    -e verilog \
     --infer-rw $(MODEL) \
     --repl-seq-mem -c:$(MODEL):-o:$(generated_dir)/$*.conf \
     -faf $(generated_dir)/$*.anno.json \
```

closed 2020-11-07 by sequencer