chipsalliance/rocket-chip #2709: What is the meaning of HellaCacheFactory?

jamesningd, 2020-11-06:
Hi all,
I want to change a little for the DCache IO port, however, I can't understand the IO define of DCache.
The sentence "new DCache(tile.hartId, tile.crossing)(p)" is how to new it, but I can't see any IO port there.
Anyone can tell me the meaning of this and how to add new IO in DCache? 

object HellaCacheFactory {
  def apply(tile: BaseTile)(p: Parameters): HellaCache = {
    if (tile.tileParams.dcache.get.nMSHRs == 0)
      new DCache(tile.hartId, tile.crossing)(p)
    else
      new NonBlockingDCache(tile.hartId)(p)
  }
}

sequencer, 2020-11-09:
`HellaCacheFactory` consumes `Tile`, based the DCache `nMSHRs` parameter to decide initiating a blocking or none blocking cache.

As for adding IOs, you should add them to `HellaCacheBundle`, which is IO between `DCache` or `NonBlockingDCache` to core, but modification to rocket-chip is not supported by issue, you can try to add them, and use it on your own risk,

closed 2020-11-09 by sequencer

jerryhethatday, 2022-02-07:
> `HellaCacheFactory` consumes `Tile`, based the DCache `nMSHRs` parameter to decide initiating a blocking or none blocking cache.
>
> As for adding IOs, you should add them to `HellaCacheBundle`, which [...]

Could you please clarify what MSHR is?

ly0, 2022-02-07:
@jerryhethatday  It's **M**icro**S**oft's **H**uman**R**esource!

sequencer, 2022-02-07:
Miss Status Handling Register
@ly0 please…

jerryhethatday, 2022-02-07:
@sequencer  Thanks. When nMSHRs  === 0,  the cache is non blocking.  Does nMSHR indicate  number of misses to be outstanding in the cache before blocking?

jamesningd, 2022-02-07:
> 

Hi Jerry,
It is a nonblocking Cache, thank you

jamesningd, 2022-02-08:
> Could you please clarify what MSHR is?

> > 
> 
> Hi Jerry, It is a nonblocking Cache, thank you

Hi Jerry
Sorry， It is blocking Cache, please neglect the first reply.