chipsalliance/rocket-chip #2733: Mem-AXI malfunctions

K16DIABLO, 2020-11-17:
I'm building a shell on VC707 (not _sifive/freedom_ repo). My system works well when the address map can be represented in 32-bit. 
[image]

However, if not, it fails to copy the BBL to DRAM during FSBL.
[image]

I just modified parameters in _src/main/subsystem/Config.scala_. Should I modify other files?

mwachs5, 2020-11-17:
Can you give more insight into what "fails to copy" means? What symptoms are you observing and how is it different from what you expect?

K16DIABLO, 2020-11-19:
FSBL should copy BBL from SD card to DRAM. Destination address is 0xBF00_0000 ~ 0xBFFF_FFFF in both cases. Since the destination address is unchanged, FSBL should have copied BBL. Only DRAM access is not successful. UART shows _printf_ successfully.

sequencer, 2020-11-19:
Here are two ideas to debug this issue:
1. Connect your SoC with OpenOCD, enable SBA to access DRAM. If failed, SoC interconnection configuration should be checked, including your AXIMem configuration, FPGA Memory controller(MIG) configuration, FPGA Memroy Phy configuration.
2. Connect your SoC with OpenOCD, disable SBA to access DRAM, If failed, Rocket configuration should be checked.
3. If OpenOCD behaves correct, you need to check your program and software toolchains.