chipsalliance/rocket-chip #2735: two each, a typo

jerryhethatday, 2020-11-18:
https://github.com/chipsalliance/rocket-chip/blob/c28c051f85e0118195749549939be58ef9bfd19a/src/main/scala/diplomacy/package.scala#L62

mwachs5, 2020-11-18:
thanks! feel free to open a PR to fix this!

jerryhethatday, 2020-11-25:
> thanks! feel free to open a PR to fix this!

Sorry about the late reply, I am busy reading thru the rocketchip codebase, I found some other typos, maybe I can open  a PR to fix all these after finishing the diplomacy source code part. 
Also, I wanna know if there is something like question template with which I can post my question on the codebase, so that I can get a quicker response, I post lots of problems recently, and did not get any reply, I wonder if I did something wrong???
Thanks for your harwork!

closed 2021-05-26 by jerryhethatday