chipsalliance/rocket-chip #2766: verilog.c:308:10: fatal error: verilog.h: No such file or directory

KrispyCamel4u, 2020-12-22:
**Type of issue**: verilog.h: such file or directory

**Development Phase**: request 

**If the current behaviour is a bug, please provide the steps to reproduce the problem:**
I cloned the repo and tried the following commands:
cd emulator
make

**What is the current behaviour?**
In file included from ../V3ParseGrammar.cpp:27:
verilog.c:308:10: fatal error: verilog.h: No such file or directory
compilation terminated.

**Please tell us about your environment:**
OS: Pop 20.10 groovy
Kernel: x86_64 Linux 5.8.0-7630-generic
Shell: bash 5.0.17

deepakpixel, 2020-12-25:
Make sure you install all the required package dependencies and rocket toolchain. Hopefully, it should work.

KrispyCamel4u, 2020-12-25:
All packages are at their newest versions but nothing seems to work.
PS: Rocket-toolchain was successfully built.

deepakpixel, 2020-12-26:
I came across this [https://github.com/verilator/verilator/pull/2505](url) check if it helps. Although the pull request has been merged and was supposed to fix the issue, you can still try downgrading or upgrading the bison version.

KrispyCamel4u, 2020-12-26:
Thank you, it did help.
I had Bison 3.7._(the newest that can be found on ubuntu 20.10) which is getting installed by default, had a problem.
According to this [https://github.com/verilator/verilator/pull/2505](url), the problem should've been fixed but didn't work in my case. 
I installed Bison 3.5.1, the default package version in ubuntu 20.04, which solved the problem.

For those who are still facing the issue:
Make sure to install flex and other required packages with Bison 3.5.1
The previous versions[3.5.1] of Bison [installation guide]:[https://geeksww.com/tutorials/miscellaneous/bison_gnu_parser_generator/installation/installing_bison_gnu_parser_generator_ubuntu_linux.php](url)

closed 2020-12-26 by KrispyCamel4u