chipsalliance/rocket-chip #2768: How can I shared a FPU between two Roceket cores？

leizhangpku, 2020-12-28:
**Type of issue**: bug report | feature request | other enhancement

**Impact**: no functional change | API addition (no impact on existing code) | API modification | unknown

**Development Phase**: request | proposal

**Other information**
 

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**

**What is the expected behavior?**

**Please tell us about your environment:**

**What is the use case for changing the behavior?**

hyf6661669, 2020-12-30:
I think maybe you could use FPU as a peripheral. The instructions which need to use FPU should send a "req" to it, which means you need to implement an arbiter.

jerryz123, 2020-12-30:
You can share the FP functional units between two tiles (cores). Note that the FPU module contains FP registers in addition to the FP functional units, so it cannot be shared across tiles.

The challenge with sharing the functional units is piping the signals from one tile to the other. This can probably most easily be implemented with Chisel BoringUtils. In any case, this will require substantial modification to the FPU module, to force one tile to use another tile's FP functional units.