chipsalliance/rocket-chip #2772: AddressSet confusion on the mask?

jerryhethatday, 2021-01-05:
The comment of `case class AddressSet` indicates that `the Base is the base address, and mask are the bits consumed by the manager`, and I just feel confused about the definition of mask. In general, an adress range can be described as the base + offset, I thought the mask should be identical to the offset, but the example tells another story: `base=0x1000, mask=0xf0f decribes a device managing 0x1000-0x100f, 0x1100-0x110f, ...` This means that the tuple of (base, mask) represents two memory spaces :`0x1000-0x100f` and `0x1100-0x110f`, I wonder what the specific computation process of this memory space is? Can anyone clarify this for me? Thanks

https://github.com/chipsalliance/rocket-chip/blob/66a60e55a18fae859730b01c04450b2da3630e19/src/main/scala/diplomacy/Parameters.scala#L128

lsteveol, 2021-01-15:
This took me a while to grasp as well. The following thread is pretty decent though
https://github.com/chipsalliance/rocket-chip/issues/1579

I have in general been using `AddressSet.misaligned(base, size)` which will create multiple AddressSets to fill a range.

Johnmc104, 2021-03-10:
> This took me a while to grasp as well. The following thread is pretty decent though
> #1579
> 
> I have in general been using `AddressSet.misaligned(base, size)` which will create multiple AddressSets to fill a range.

when I try to add CanHaveMasterAHBMMIOPort， i set the address
`address       = AddressSet.misaligned(0x20000000, 0x00400000),`

`Interrupt map (2 harts 2 interrupts):
  [1, 2] => ldut

 java.lang.IllegalArgumentException: requirement failed: SimpleBus mmio-port-ahb must set ranges`

the second argument, Isn't it a range? Thanks

lsteveol, 2021-03-10:
I'm assuming `CanHaveMasterAHBMMIOPort` is something you've created, but likely similar to the `CanHaveMasterAXI4MMIOPort`. The error you are seeing is due to a `require`ment on ranges not being empty. 

Are you setting the `resources` in the `AXI4SlaveParameters`?

```scala
trait CanHaveMasterAXI4MMIOPort { this: BaseSubsystem =>
  private val mmioPortParamsOpt = p(ExtBus)
  private val portName = "mmio_port_axi4"
  private val device = new SimpleBus(portName.kebab, Nil)

  val mmioAXI4Node = AXI4SlaveNode(
    mmioPortParamsOpt.map(params =>
      AXI4SlavePortParameters(
        slaves = Seq(AXI4SlaveParameters(
          address       = AddressSet.misaligned(params.base, params.size),
          resources     = device.ranges,   <-----
          executable    = params.executable,
          supportsWrite = TransferSizes(1, params.maxXferBytes),
          supportsRead  = TransferSizes(1, params.maxXferBytes))),

```

Johnmc104, 2021-03-11:
> I'm assuming `CanHaveMasterAHBMMIOPort` is something you've created, but likely similar to the `CanHaveMasterAXI4MMIOPort`. The error you are seeing is due to a `require`ment on ranges not being [...]

it works！ Thank you very much.

jerryhethatday, 2021-04-18:
> This took me a while to grasp as well. The following thread is pretty decent though
> #1579
> 
> I have in general been using `AddressSet.misaligned(base, size)` which will create multiple AddressSets to fill a range.

Thanks  for your reply, And I get the general idea of AddressSet. I have another confusion, can you clarify the `widen` method of AddressSet:
`
  // Widen the match function to ignore all bits in imask
  def widen(imask: BigInt) = AddressSet(base & ~imask, mask | imask)`
What does this method do? what does this method widen? Thanks

lsteveol, 2021-04-18:
I'm not 100% sure as I haven't used this before, but it appears to return an `AddressSet` that includes a new "mask" (meaning new size) taken into account. If you look at `tielink/RegisterRouter.scala`, you can see this `widen` method used for the `TLRegisterNode`. In the case of the `TLRegisterRouter` it appears to be used as a check that all of the `AddressSet`s are defined properly.

Here is a small example I did. I have an AddressSet defined for a `ReigsterRouter`
```scala
val node = TLRegisterNode(
    address = AddressSet.misaligned(baseAddr, 0x100),  //baseAddr here we can assume is 0x0
    device  = device,
    concurrency = 1, 
    beatBytes = 4)

  node.address.foreach{ case a =>
    println(s"AddressSet.widen by x100: ${a.widen(0x100)}")
    println(s"AddressSet.widen by x200: ${a.widen(0x200)}")
  }
```
And the following is printed
```
AddressSet.widen by x100: AddressSet(0x0, 0x1ff)
AddressSet.widen by x200: AddressSet(0x0, 0x2ff)
```

One thing I believe you could do with this is if you are adding ports/nodes to a diplomacy graph, a `NexusNode` such as a `TLXbar` could use this to "`widen`" the slave port(s) to calculate new addressing requirements. And/or if you have a particular block in which your `RegisterRouter` node can add/remove registers based on some parameters, you can calculate the new `AddressSet` _after_ the node creation instead of before. These are obviously more complex operations.

jerryhethatday, 2021-04-25:
> I'm not 100% sure as I haven't used this before, but it appears to return an `AddressSet` that includes a new "mask" (meaning new size) taken into account. If you look at [...]

Thanks for your reply. I get the general idea of widen, I will close this issue. But I still felt very confused about the widen usage in this new issue: https://github.com/chipsalliance/rocket-chip/issues/2828  it will be so cool if you can clarify that. Thanks! I think the AddressSet is cool

closed 2021-04-25 by jerryhethatday