chipsalliance/rocket-chip #2793: Problem in generating make files

SaadUllah-spec, 2021-03-15:
https://reference.digilentinc.com/reference/programmable-logic/arty-a7/arty_a7_100_risc_v/start

I am trying to implement the above link to learn about RISC-V. However when I try ./build.sh I get a bunch of errors. The makefile errors are what I am more concerned about anyone with a solution help me out!!
usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/saad/freedom/rocket-chip/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/saad/freedom/rocket-chip/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here

collect2: error: ld returned 1 exit status

gmake[2]: *** [Makefile:2897: src/openocd] Error 1

gmake[1]: *** [Makefile:4181: all-recursive] Error 1

gmake: *** [Makefile:1953: all] Error 2

pcotret, 2022-03-07:
Got a similar error, solution in my case had two steps:
- Checkout the version noted there: https://github.com/chipsalliance/rocket-chip/blob/master/README.md?plain=1#L35-L36
- Using a GCC9 instead of my GCC 10 :
```bash
https://github.com/chipsalliance/rocket-tools
cd rocket-tools
git checkout e2c6d1577a75f506fe992c3eb20a75174504476e
git submodule update --init --recursive
export RISCV=/path/to/install/riscv/toolchain
export MAKEFLAGS="$MAKEFLAGS -jN" # Assuming you have N cores on your host system
GCC=gcc-9 CXX=g++-9 ./build.sh
GCC=gcc-9 CXX=g++-9 ./build-rv32ima.sh (if you are using RV32).
```

pcotret, 2022-03-07:
Got everything compiled with GCC 9:
```bash
git clone https://github.com/freechipsproject/rocket-tools
cd rocket-tools
git submodule update --init --recursive
export RISCV=/path/to/install/riscv/toolchain
export MAKEFLAGS="$MAKEFLAGS -jN" # Assuming you have N cores on your host system
$ CC=gcc-9 CXX=g++-9 ./build.sh
$ CC=gcc-9 CXX=g++-9 ./build-rv32ima # For 32-bit
```

closed 2024-07-24 by michael-etzkorn