chipsalliance/rocket-chip #2799: Confusion on TLBufferNode's clientFn and managerFn

jerryhethatday, 2021-03-29:
The TLBufferNode is an adpter node, which only modifies the parameter pass thru without changing the number of ports.  
The `D=>D , and U=> U` in TLBufferNode situation is as below:
`
    clientFn  = { p => p.v1copy(minLatency = p.minLatency + b.latency + c.latency) },
    managerFn = { p => p.v1copy(minLatency = p.minLatency + a.latency + d.latency) }
`
I have confusion on this. Why the latency for the masterport after going thru this node becomes `p.minLatency + b.latency + c.latency`, mostly master will initiate a message from Channel A and got the ACK from chanel D, so I think the latency after going thru the buffer node should just be  `p.minLatency + a.latency + d.latency + e.latenct(for some cases)`. similarly the slave(manager) latency after going thru the buffer node should be  `p.minLatency + b.latency + c.latency`
Can anyone clarify this? I know I got something wrong, but It is so hard for me now to go thru the specific detail. 
Thanks!
https://github.com/chipsalliance/rocket-chip/blob/66a60e55a18fae859730b01c04450b2da3630e19/src/main/scala/tilelink/Buffer.scala#L15

jerryhethatday, 2021-03-29:
@hcook

hcook, 2021-05-06:
`clientFn` is the view of the clients as seen by the managers. If a manager sends a B probe request to a client and receives a C response, the latency of this transaction will be b.latency + c.latency. `managerFn` is the view of the managers as seen by the clients. If a client sends an A request to a manager a receives a D response the total latency of this transaction will be a.latency + d.latency.  In general clients don't care about e.latency because they can fire the acknowledgement and forget about it, they don't have to wait until it is received before issuing their next request.

hcook, 2021-05-06:
Also not that these are all _minimum_ latency, i.e. the fastest possible transaction that could ever occur, not the expected latency or average latency or maximum latency. Most often this field is just used for checking whether the value is 0 or not 0: 0 means that the other agent is responding in the same cycle the request was received, which the TL spec allows but not all agents wish to support.

closed 2021-06-11 by jerryhethatday