chipsalliance/rocket-chip #2800: Comment Error

jerryhethatday, 2021-03-29:
The comment says that `flip or not in [[AutoBundle.makeElements]]. If true this corresponds to danglesOut, if false it corresponds to danglesIn`. IMHO, this is obviously wrong:
```
  protected[diplomacy] def danglesOut: Seq[Dangle] = oPorts.zipWithIndex.map { case ((j, n, _, _), i) =>
    Dangle(
      source = HalfEdge(serial, i),
      sink   = HalfEdge(n.serial, j),
      flipped= false,
      name   = wirePrefix + "out",
      data   = bundleOut(i))
  }
```

https://github.com/chipsalliance/rocket-chip/blob/fe1ba112be07b0c78db9cb9019f032d9d751c4fe/src/main/scala/diplomacy/LazyModule.scala#L476

jerryhethatday, 2021-05-11:
@richardxia @mwachs5

richardxia, 2021-05-11:
Hi @jerryhethatday, I unfortunately can't answer your question, since I don't actually know much about this code, and I'm not actively work on it. The rocket-chip new issue template says that support questions may be asked on Stack Overflow using the \[rocket-chip\] tag, https://stackoverflow.com/questions/tagged/rocket-chip, so you might have better luck getting your questions answered there.

jerryhethatday, 2021-05-12:
@richardxia  Thanks Richard, I felt like this is certainly a comment error. Anyway, I will close this and  post this on stackoverflow. Thanks again for your quick reply.

closed 2021-05-12 by jerryhethatday