chipsalliance/rocket-chip #2801: make verilog error!

jiejie324, 2021-03-29:
cd /home/lj/rocket-chip && java -Xmx2G -Xss8M -jar /home/lj/rocket-chip/sbt-launch.jar -Dsbt.sourcemode=true -Dsbt.workspace=$PWD assembly
[info] welcome to sbt 1.3.13 (Ubuntu Java 11.0.10)
[info] loading settings for project rocket-chip-build from plugins.sbt ...
[info] loading project definition from /home/lj/rocket-chip/project
[info] loading settings for project rocketchip from build.sbt ...
[info] loading settings for project hardfloat from build.sbt ...
[info] loading settings for project chisel3-build from plugins.sbt ...
[info] loading project definition from /home/lj/rocket-chip/chisel3/project
[info] loading settings for project chisel from build.sbt ...
[info] loading settings for project firrtl-build from plugins.sbt ...
[info] loading project definition from /home/lj/rocket-chip/firrtl/project
[info] loading settings for project firrtl from build.sbt ...
[info] resolving key references (14162 settings) ...
Using addons: 
[info] set current project to rocketchip (in build file:/home/lj/rocket-chip/)
[info] Compiling 376 Scala sources to /home/lj/rocket-chip/target/scala-2.12/classes ...
[error] /home/lj/rocket-chip/src/main/scala/package.scala:6:16: not found: value chipsalliance
[error]   val config = chipsalliance.rocketchip.config
[error]                ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/AHBLite.scala:9:29: not found: type Parameters
[error] class AHBLite()(implicit p: Parameters) extends LazyModule {
[error]                             ^
[error] /home/lj/rocket-chip/src/main/scala/diplomacy/LazyModule.scala:28:45: not found: type Parameters
[error] abstract class LazyModule()(implicit val p: Parameters) {
[error]                                             ^
[error] /home/lj/rocket-chip/src/main/scala/diplomacy/LazyModule.scala:269:19: not found: type Parameters
[error]   implicit val p: Parameters = wrapper.p
[error]                   ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/Parameters.scala:127:11: not found: type Parameters
[error]   params: Parameters,
[error]           ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/AHBLite.scala:38:27: not found: type Parameters
[error]   def apply()(implicit p: Parameters) = {
[error]                           ^
[warn] /home/lj/rocket-chip/src/main/scala/amba/ahb/AHBLite.scala:6:36: Unused import
[warn] import freechips.rocketchip.config.Parameters
[warn]                                    ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/Nodes.scala:10:44: not found: type Field
[error] case object AHBSlaveMonitorBuilder extends Field[AHBSlaveMonitorArgs => AHBSlaveMonitorBase]
[error]                                            ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/Nodes.scala:15:72: not found: type Parameters
[error]   def edge(pd: AHBMasterPortParameters, pu: AHBSlavePortParameters, p: Parameters, sourceInfo: SourceInfo) = AHBEdgeParameters(pd, pu, p, sourceInfo)
[error]                                                                        ^
[error] /home/lj/rocket-chip/src/main/scala/diplomacy/Nodes.scala:173:29: not found: type Parameters
[error]   def edge(pd: D, pu: U, p: Parameters, sourceInfo: SourceInfo): E
[error]                             ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/Nodes.scala:32:45: not found: type Field
[error] case object AHBMasterMonitorBuilder extends Field[AHBMasterMonitorArgs => AHBMasterMonitorBase]
[error]                                             ^
[error] /home/lj/rocket-chip/src/main/scala/amba/ahb/Nodes.scala:37:72: not found: type Parameters
[error]   def edge(pd: AHBMasterPortParameters, pu: AHBSlavePortParameters, p: Parameters, sourceInfo: SourceInfo) = AHBEdgeParameters(pd, pu, p, sourceInfo)
[error]

jiejie324, 2021-03-29:
please help me!

colinschmidt, 2021-03-29:
You need to initialize the git submodules

jiejie324, 2021-03-29:
> You need to initialize the git submodules

Thank you for your reply.Already Initialize git submodule，it's the same error.Do you have any other ideas？

zk-qiang, 2021-04-12:
vsim folder make verilog error! I have the same problem. Have you solved this problem?

jiejie324, 2021-04-12:
> vsim文件夹使verilog错误！我也有同样的问题。你解决了这个问题吗？

Has been resolved

jiejie324, 2021-04-12:
> vsim文件夹使verilog错误！我也有同样的问题。你解决了这个问题吗？

I am in China, after over the wall, following the tutorial operation is successful

zk-qiang, 2021-04-13:
I am also in China, may I ask what tool you use over the wall-------- Original message --------From: jiejie324 ***@***.***>Date: Tue, Apr 13, 2021, 01:04To: chipsalliance/rocket-chip ***@***.***>Cc: zk-qiang ***@***.***>, Comment ***@***.***>Subject: Re: [chipsalliance/rocket-chip] vsim folder make verilog error! (#2801)

vsim文件夹使verilog错误！我也有同样的问题。你解决了这个问题吗？

I am in China, after over the wall, following the tutorial operation is successful

sequencer, 2021-04-13:
Follow readme to initiate git submodule should resolve.

closed 2021-04-13 by sequencer

zk-qiang, 2021-04-13:
Please provide your email address. I have some questions to consult you-------- Original message --------From: jiejie324 ***@***.***>Date: Tue, Apr 13, 2021, 01:04To: chipsalliance/rocket-chip ***@***.***>Cc: zk-qiang ***@***.***>, Comment ***@***.***>Subject: Re: [chipsalliance/rocket-chip] vsim folder make verilog error! (#2801)

vsim文件夹使verilog错误！我也有同样的问题。你解决了这个问题吗？

I am in China, after over the wall, following the tutorial operation is successful

jiejie324, 2021-04-13:
2638469667@qq.com