chipsalliance/rocket-chip #2807: How are parent and child lazymodule's clock and reset are connected?

jerryhethatday, 2021-04-08:
Every chisel module has implict clock and reset, I wonder if the parent and child's clock and reset are automatically connected?
Also, I wonder if parent and child **lazymodule's** clock and reset are connected by default(considering the module property of Lazymodule is LazyModuleImp, I know if the module of LazyModule is of type LazyRawModuleImp, then you can wire the child clock and child reset manully. I just felt confused about the default behavior)? If they are connected, which part of the code base indicate this? 
Thanks!
Part of this is a chisel question, So @jackkoenig, can you clarify this for me?        
@hcook @terpstra

jackkoenig, 2021-04-08:
> Every chisel module has implict clock and reset

Not strictly *every* module, `Module` and `MultiIOModule` have implicit clock and reset, but `RawModule` does not have implicit clock nor reset.

> I wonder if the parent and child's clock and reset are automatically connected?

For `Module` and `MultiIOModule`, this is correct.

This behavior is described (albeit, less clearly than it could be) here: https://www.chisel-lang.org/chisel3/docs/explanations/modules.html

I believe the main logic handling this sort of thing in Diplomacy is here (https://github.com/chipsalliance/rocket-chip/blob/025184e1a0fdd8338eb0f66ad9ab33587b619854/src/main/scala/diplomacy/LazyModule.scala#L350).`LazyModuleImp` is a `MultiIOModule` and `LazyRawModuleImp` is (you guessed it) a `RawModule` so the behavior somewhat matches there.

The main difference is that, while in standard Chisel module hierarchies you can use `withClock`, `withReset` and `withClockAndReset` to change the implicit clocks and resets for the nested scope (see https://www.chisel-lang.org/chisel3/docs/explanations/multi-clock.html), this doesn't work for Diplomacy because the `LazyModule` hierarchy lives outside of the Chisel pieces (ie. `LazyModuleImps`). Instead, you can drive `childClock` and `childReset` to provide default values for all `LazyModuleImp` children, and you can override specific clocks and resets by directly connecting to the `.clock` and `.reset` inputs on any child modules.

andguilty, 2021-04-15:
Their LazymoduleImp‘s clock and reset are connect in fact。Lazymodule is not a hardware module。

andguilty, 2021-04-16:
You can have a look at function ```MultiIOModule.initializeInParent``` and object ```Module```'s  ```apply``` function. the connection of parent clock and children clock is specific .