chipsalliance/rocket-chip #2812: TLBs present even if useVM=false

Uzair540, 2021-04-12:
There is a parameter in case class `RocketCoreParams` that is `useVM` . When I disable that it should not use TLBs as TLBs are used for translation of virtual address to physical address. However, when I see the generated verilog, it contains TLBs in DCache as well as FrontEnd. Am I missing something?

jerryz123, 2021-04-12:
If `!usingVM`, the TLBs contains very little logic, and just pass through PPN = VPN.

Uzair540, 2021-04-13:
Can you please elaborate PPN=VPN ?

bg193, 2021-04-23:
> If `!usingVM`, the TLBs contains very little logic, and just pass through PPN = VPN.

When I disable useVM, I see the PTW in the generated verilog. How to disable PTW?

jerryz123, 2021-04-23:
PTW module is still instantiated in that case but most internal logic is removed.