chipsalliance/rocket-chip #2823: problem on src/diplomacy/adder_tutorial.md

YuanZheCN, 2021-04-19:
The code here is not right?
[image]

The buildOutput says that:
value io is not a member of freechips.rocketchip.diplomacy.LazyModuleImp
    when(monitor.module.io.error) {

YuanZheCN, 2021-04-20:
setup.bst need to change

closed 2021-04-20 by YuanZheCN

zhuanhao-wu, 2022-02-13:
Hi @YuanZheCN, I'm facing the same issue and I cannot access `monitor.module.io.error`, I wonder if you could suggest a way to make it work?

YingXiangLi10, 2022-11-15:
@zhuanhao-wu 
I encountered the same problem. But After I changed my chisel version, it worked. Maybe it only works using chisel 3.3.3. Here's my build.sbt
```
ThisBuild / scalaVersion     := "2.12.12"
ThisBuild / version          := "0.1.0"

lazy val root = (project in file("."))
  .settings(
    name := "chisel_template",
    libraryDependencies ++= Seq(
      "edu.berkeley.cs" %% "chisel3" % "3.3.3",
      "edu.berkeley.cs" %% "chiseltest" % "0.3.1" % "test",
      "edu.berkeley.cs" %% "rocketchip" % "1.2.+"

    ),
    scalacOptions ++= Seq(
      "-Xsource:2.11",
      "-language:reflectiveCalls",
      "-deprecation",
      "-feature",
      "-Xcheckinit"
    ),
    addCompilerPlugin("org.scalamacros" % "paradise" % "2.1.1" cross CrossVersion.full)
  )
```