chipsalliance/rocket-chip #2827: BTB does not use strongly_taken defined inside BHTResp bundle

hadirkhan10, 2021-04-23:
In the `BTB.scala` file there is a bundle:

```scala
class BHTResp(implicit p: Parameters) extends BtbBundle()(p) {
  val history = UInt(width = btbParams.bhtParams.map(_.historyLength).getOrElse(1))
  val value = UInt(width = btbParams.bhtParams.map(_.counterLength).getOrElse(1))
  def taken = value(0)
  def strongly_taken = value === 1
}
```

I searched the `strongly_taken` method and it is not being used anywhere inside the file. How is this handling being done?

a0u, 2021-04-23:
`strongly_taken` is used in the Frontend to determine whether to update the BTB:

https://github.com/chipsalliance/rocket-chip/blob/86a2f2cca699f149bcc082ef2828654a0a4e3f4b/src/main/scala/rocket/Frontend.scala#L273-L275

hadirkhan10, 2021-04-23:
Oh I see @a0u . I am assuming `counterLength` identifies the predictor bits to use. By default it is 1 which means (Taken, Not Taken). If we set it to 2 it will then use `Weakly Taken`, `Strongly Taken`, `Weakly Not Taken` and `Strongly Not Taken`?
I am a little confused with these two statements:
```
def taken = value(0)
def strongly_taken = value === 1
```

a0u, 2021-04-23:
The least-significant bit of the counter yields the predicted direction.

The two-bit "counter" differs from the usual textbook encoding of a 2-bit saturating counter.  The `0b01` state corresponds to "strongly taken"; the state machine is defined here:
https://github.com/chipsalliance/rocket-chip/blob/86a2f2cca699f149bcc082ef2828654a0a4e3f4b/src/main/scala/rocket/BTB.scala#L104

hadirkhan10, 2021-04-29:
okay thanks

closed 2021-04-29 by hadirkhan10