chipsalliance/rocket-chip #2837: What's the use of  user and echo in TLBundle A C D channel？

ddxoxpp, 2021-05-15:
I notice that there are two BundleMap named ‘’‘user’‘’ and ‘’‘echo’‘’ which aren‘t mentioned in Tilelink specification . I want to know the purpose of them but there is no document about them. So , What's the use of  '''user''' and '''echo''' in TLBundle A C D ?channel？

mwachs5, 2021-05-15:
User bits are extra fields included for user-defined purpose. Echo bits are also extra fields included for user-defined purpose, but must be "echo-ed" back in the response.

ddxoxpp, 2021-05-16:
> User bits are extra fields included for user-defined purpose. Echo bits are also extra fields included for user-defined purpose, but must be "echo-ed" back in the response.

So in Rocketchip template ， they are not used actually？

mwachs5, 2021-05-16:
The rocket-chip repo is a library of utilities for SoC design. there are many available features that are not used in the repo itself.

closed 2021-05-16 by ddxoxpp