chipsalliance/rocket-chip #2840: Run out of memory when add >300 vector instructions

huayue-liang, 2021-06-01:
**Type of issue**: bug report

**Impact**:  unknown

**Development Phase**: proposal

**Other information**
 
I add all vector instructions in the rocket/Instructions.scala files to decode table(rocket/IDecode.scala, rocket/RocketCore.scala),
`(usingVector.option(new VECDecode)) ++:`

When in compile stage `| => chipyard / Compile / runMain 1601s`, it takes a lot of times and exits with an error: 
`Java ... Run Out of Memory ... `.

It does work when add less vector instructions, for example about 100 VFPU instuctions.

I run "make" in the chipyard project environment: "chipyard/sims/verilator".

**If the current behavior is a bug, please provide the steps to reproduce the problem:**

**What is the current behavior?**
I have tried some config, but it does not work:

1. add memory size from "Xmx8G" to "Xmx32G"
2. divide the vector instructions into different group and decode them in different modules:

```
class VMEMDecodeT(implicit val p: Parameters) extends DecodeConstants {
+-- 50 lines: val table: Array[(BitPat, List[BitPat])] = Array(---------------------------------------
}

class VFPUDecodeT(implicit val p: Parameters) extends DecodeConstants {
+--110 lines: val table: Array[(BitPat, List[BitPat])] = Array(---------------------------------------
}
...

class VPUImp(outer: VPU)(implicit p: Parameters) extends LazyModuleImp(outer) {
  val id_ctrl_mem = Module(new VMEMImp)
  val id_ctrl_fpu = Module(new VFPUImp)
  val id_ctrl_vxx = Module(new VVXXImp)
  val id_ctrl_vvx = Module(new VVVXImp)
  ...
}
```

**What is the expected behavior?**

**Please tell us about your environment:**

- chipyard: commit cca3cd087c1c5a54f98867bc4a89129c1c299b94
- rocket-chip: commit 00d9e02c1901d0285f22eefec7c25073fb90e4f1
- OS: Ubuntu 18.04.5 LTS  Linux 4.15.0-141-generic  x86-64

**What is the use case for changing the behavior?**

sequencer, 2021-06-01:
From your description, I think you are running into the QMC np-hard algorithm issue.
Which will be resolved in chipsalliance/chisel3#1914,

jackkoenig, 2021-06-03:
While it is an NP-hard problem, this can generally be solved by splitting up the code problem into smaller problems.

You said:

> It does work when add less vector instructions, for example about 100 VFPU instuctions.

and 

>  have tried some config, but it does not work:
>  1. ...
>  2. divide the vector instructions into different group and decode them in different modules:

These aren't compatible with each other, if it can work for 100 instructions then it should work fine if you split up the decoding. It's less about putting things in different modules than it is about reducing how much you pass to `DecodeLogic`.

https://github.com/chipsalliance/chisel3/pull/1914 will be better, but you should be able to work around this by splitting up the problem.

closed 2021-09-26 by huayue-liang