chipsalliance/rocket-chip #2852: How to connect LazyModule's clock with its Tilelink node clock?

gkaplayan, 2021-06-30:
It is possible to connect explicit clocks for Module, RawModule and MultiIOModule by using withClock. Additionally, you can also connect explicit clock for a LazyModule by using childClock. If the instantiated LazyModule has a Tilelink Node. How can I connect its clock to the Tilelink Node's Clock?

For example There is a top level LazyModule which instantiates two LazyModules with Tilelink Nodes. The Tilelink Nodes have different clocks. How can I implement this scenario?

The issue is similar to https://github.com/chipsalliance/rocket-chip/issues/2807. Could you please clarify my issue for me, @jackkoenig?

lsteveol, 2021-07-09:
When you have a LazyModule, there is a `module` method/variable that you can use to reference the actual hardware. From there you can set the `clock`/`reset` to whatever you would like.

jackkoenig, 2021-07-09:
Is the hierarchy something like this?
```scala
class MyLazyModule extends LazyModule {
  val node = new TileLinkNode

  // And you want to change the clock in here?
  lazy val module = new LazyModuleImp {
  }
}
```
I think you use `withClock` inside `MyLazyModule's` `LazyModuleImp`:
```scala
class MyLazyModule extends LazyModule {
  val node = new TileLinkNode

  lazy val module = new LazyModuleImp { withClock(node.getClockSomehow) {
  } }
}

gkaplayan, 2021-07-11:
My hierarchy is like this:
`class MyTopModule extends LazyModule {
  val lazy1 = LazyModule(new MyLazyModule1())
  val lazy2 = LazyModule(new MyLazyModule2())
	
  lazy val module = new LazyModuleImp {
    val io = IO(new Bundle {
      val clk_2 = Input(Clock())
    }	    
  // lazy2.clock := io.clk_2 // gives a Chisel error
  // WithClock(clk_2){ val lazy2 = Module(LazyModule(new MyLazyModule2()).module)} // cannot reach node2
  }
}

class MyLazyModule1 extends LazyModule {
  val node1 = new TileLinkNode

  lazy val module = new LazyModuleImp {}
}

class MyLazyModule2 extends LazyModule {
  val node2 = new TileLinkNode

  lazy val module = new LazyModuleImp {}
}`

As shown below with comments, my two solution does not work for me. 

@jackkoenig I need something similar to node.getClock(). I am also open to your proposal about the hierarchy. My goal is to desing a IP with two nodes whose clocks are different. 

@lsteveol Could you give me an example for my example code above?

Thank you both @jackkoenig and @lsteveol for your help

lsteveol, 2021-07-11:
What I was referring to would have been something like this

```scala
class MyTopModule extends LazyModule {
  val lazy1 = LazyModule(new MyLazyModule1())
  val lazy2 = LazyModule(new MyLazyModule2())
	
  lazy val module = new LazyModuleImp {
    val io = IO(new Bundle {
      val clk_2 = Input(Clock())
    }	    
    lazy2.module.clock := io.clk_2 // should work
  }
}
```

Although, being able to get the clock from the node like Jack is suggesting would likely be a much better solution.

gkaplayan, 2021-07-12:
It worked, thanks for your help @lsteveol. As a feature request, it would be good to have node.getClock().

Let me share the final example code and close the issue.
```
class MyTopModule extends LazyModule {
  val lazy1 = LazyModule(new MyLazyModule1())
  val lazy2 = LazyModule(new MyLazyModule2())
	
  lazy val module = new LazyModuleImp {
    val io = IO(new Bundle {
      val clk_2 = Input(Clock())
      val rst_2 = Input(Bool())
    }
  lazy2.module.clock := io.clk_2
  lazy2.module.reset := io.rst_2
  // lazy2.module.io.clock := io.clk_2 // gives a Chisel error	    
  // lazy2.clock := io.clk_2 // gives a Chisel error
  // WithClock(clk_2){ val lazy2 = Module(LazyModule(new MyLazyModule2()).module)} // cannot reach node2
  }
}

class MyLazyModule1 extends LazyModule {
  val node1 = new TileLinkNode

  lazy val module = new LazyModuleImp {}
}

class MyLazyModule2 extends LazyModule {
  val node2 = new TileLinkNode

  lazy val module = new LazyModuleImp {}
}
```

closed 2021-07-12 by gkaplayan