chipsalliance/rocket-chip #2860: rocket chip TinyConfig 32 bit emulator execution failure

Rodiot, 2021-07-29:
Hi,

I am trying to run and build the TinyConfig configuration from the standard configs within the _src/main/scala/system/Config.scala_ file.
The building and starting of the emulator with the steps described in the git repo works fine.

To build a simple _hello_ program I use the 32 bit gcc of the rocket tools and the _-march=rv32i -mabi=ilp32_ switches as the TinyConfig does not have a FPU (at least to my understanding).
Further, I use the Makescript and test.ld of the riscv-tests repo.
In the makefile, the following gcc options are used
_-DPREALLOCATE=1 -mcmodel=medany -static -std=gnu99 -O1 -Og -g -fno-common -fno-builtin-printf_
_-static -nostdlib -nostartfiles -lm -lgcc -T $(src_dir)/common/test.ld_

Now when I run the emulator with
`./emulator-freechips.rocketchip.system-TinyConfig test/hello`
I am getting the following error:

_*** FAILED *** (tohost = 1337)
*** FAILED *** via dtm (code = 1337, seed 1627614307) after 286666 cycles_

I dug deeper and used the gdb debug facilities.
There I can run the program until it jumps in the _init function defined in syscalls.c, which I also used from the riscv-tests repo.
Immediately after entering the function it jumps into the trap_entry label defined in crt.S (you now by now where I got this file from :))
From there to the `handle_trap` function and further to the `tohost_exit'

When trying everything with the DefaultConfig and the 64 bit gcc everything works just fine.
Do I have to adapt the linker script or any other file for 32 bit?
If so, how?
I am out of my depth here and google didnt turn up anything useful.

Any help or hint would be appreciated.

KR
Robert

Btw:
I am on the 1.2 release branch and use the corresponding rocket tools commit (although I also encontered the problem on the master branch)

souravmohapatra, 2022-05-09:
Hello,

I am stuck with the exact same issue as well. The 64bit works while the 32bit results in the above error.

I tried tinkering with the linker script, the crt and the syscall files but to no avail. To be precise, the code jumps to the trap handler after executing the `swsp` instruction in the _init().

I appreciate any help!

Rodiot89, 2022-05-10:
Hi souravmohapatra,

I never got around to a solution for the example programs, however my own programs work.
Make sure that the stack is not outside of the available memory.
Its set in the crt.S file

\# give each core 32KB of stack + TLS  
\#define STKSHIFT 15    
  sll a2, a0, STKSHIFT
  add tp, tp, a2
  add sp, a0, 1
  sll sp, sp, STKSHIFT
  add sp, sp, tp

souravmohapatra, 2022-05-10:
Thanks a lot @Rodiot89! That worked for me as well.

michael-etzkorn, 2022-05-10:
Yes, it seems this is an issue with running a test requiring a large stack.  By default, the TinyConfig only has a 16KB scratchpad of memory. 

See: https://github.com/chipsalliance/rocket-chip/issues/430

michael-etzkorn, 2024-07-24:
Closed as repeat of #430 which is soft-blocked by #2960.

closed 2024-07-24 by michael-etzkorn