chipsalliance/rocket-chip #2862: TLMonitor bug report

ljwljwljwljw, 2021-08-05:
**Type of issue**: bug report

**Impact**: no functional change

**Development Phase**: request

**Other information**
 

https://github.com/chipsalliance/rocket-chip/blob/148d5d2ab8295ecfc3aa99f4090e06cbc2c4d220/src/main/scala/tilelink/Monitor.scala#L799-L801

The code above seems to check the case that a client send a '*Release*' to a manager from C channel and the manager send a response message '*ReleaseAck*' from D channel to client in same cycle. In this case the manager must accept the '*Release*' request before responding '*ReleaseAck*', so if '*d_ready*' the '*c_ready*' must be true.

However, if the message in C channel is '*ProbeAck*', the '*source*' field will be useless. For example, block-inclusive-cache always set '*source*' to 0 when the message is '*ProbeAck*'. 
https://github.com/sifive/block-inclusivecache-sifive/blob/a260c905acc3645a49583f1c304f1a83025a860b/design/craft/inclusivecache/src/Scheduler.scala#L130

So is it a bug to check '*c_ready*' when the opcode carried on C channel is '*ProbeAck*' instead of '*Release*'?

yuchangyuan, 2023-01-05:
Simulation end with below message, not sure is this same issue.
> Assertion failed: ready check
>    at Monitor.scala:49 assert(cond, message)

Below is the waveform of related monitor.
[image]

jerryz123, 2023-01-05:
Does this fix your issue. https://github.com/chipsalliance/rocket-chip/pull/3195

yuchangyuan, 2023-01-05:
> Does this fix your issue. #3195

yes, the issue is fixed after apply the patch from #3195

closed 2023-01-05 by sequencer