chipsalliance/rocket-chip #2865: How about maintain a build.wake?

davidmlw, 2021-08-10:
**Type of issue**: other enhancement

**Impact**: no functional change

**Development Phase**: proposal

**Other information**
 

The Wake script is a very good way to compose different design. Wake not only provide a method to compile Scala modules from source code, but also provide a way to compose different design parts to an integrated one. Wake provides many key concepts that apply to IC design flow, such as hash chain, file output chasing, differnt EDA flow integration. It is a much better solution than some other alternatives. 

The IC industry needs not only a new RTL launguage, a new basic ISA, but also a unified EDA flow. A standard, input/output clearly, pure functional dependency, Wake-based flow is very important for the IC users to have a unified and sharable experience among different groups.

@davidmlw would offer to maintain a basic rocket-chip build.wake if the community is interested. @jackkoenig  @terpstra @sequencer

sequencer, 2021-08-11:
1. wake is a state-of-the-art build system for cross projects and build from source.
2. currently mill is a simple build system which provides a good build-from-source infrastructure, I illustrated and maintained a flow to totally build-from-source chisel+firrtl+RC+chipyard [here](https://github.com/sequencer/chipsalliance-playground)
3. sbt is a standard for scala.
So I personally think sbt+mill is good enough for rocket-chip(since RC don't depend on the non-scala projects)

I'd love to see a wake build system for RC and related build systems. However, I'm not a wake expert, so I cannot review it, so I think we should left these decision to @jackkoenig and @wsnyder to see if they are willing to review these contributions.

davidmlw, 2021-08-11:
Build.wake can do the same as item 2.

sequencer, 2021-08-11:
Yes. wake is a good build system, but I'd like to see an open-source project using wake to illustrate its usage, and @jackkoenig and @terpstra being willing to review it.