chipsalliance/rocket-chip #2867: What files should I use in FPGA implemantation?

yusuftkmn, 2021-08-17:
I have completed the rocket-chip installation. But which verilog files or files will I use when synthesizing FPGA in vivado.I used freechips.rocketchip.system.DefaultFPGAConfig.behav_srams.v
freechips.rocketchip.system.DefaultFPGAConfig.v
plusarg_reader.v
EICG_wrapper.v
SimDTM.v
But i get error in SimDTM.v .

RaphaelKlink, 2021-08-17:
Hey,

for our FPGA implementation we use the Digital Top and these 3 Files
*.top.mems.v
*.top.v
plusarg_reader.v

When you want to use the Chiptop you need the 3 verilog files from above and
EICG_wrapper.v
IOCell.v
ClockDividerN.v

yusuftkmn, 2021-08-19:
I dont have top.mems.v and .top.v .Why? .I generated rocket chip .Then I used below files
freechips.rocketchip.system.DefaultFPGAConfig.behav_srams.v
freechips.rocketchip.system.DefaultFPGAConfig.v
plusarg_reader.v
EICG_wrapper.v
SimDTM.v

I synthesized it in vivado But There are 951 I/Os, is this normal?

RaphaelKlink, 2021-08-19:
Oh yes my bad. 
I did not use this repository.
I use Chipyard (https://github.com/ucb-bar/chipyard).

This is also from UCB and uses Rocketchip as the Basis for a whole SoC.