chipsalliance/rocket-chip #2876: Do ICache use virtual address to access the ITIM?

jerryhethatday, 2021-09-08:
I am reading thru the ICache code base, and felt very confused about ITIM part. See code below:
```
  for (i <- 0 until nWays) {
    val s1_idx = index(s1_vaddr, io.s1_paddr)
    val s1_tag = io.s1_paddr >> pgUntagBits
    val scratchpadHit = scratchpadWayValid(i) &&
      Mux(s1_slaveValid,
        lineInScratchpad(scratchpadLine(s1s3_slaveAddr)) && scratchpadWay(s1s3_slaveAddr) === i,
        addrInScratchpad(io.s1_paddr) && scratchpadWay(io.s1_paddr) === i)
    val s1_vb = vb_array(Cat(UInt(i), s1_idx)) && !s1_slaveValid
    val enc_tag = tECC.decode(tag_rdata(i))
    val (tl_error, tag) = Split(enc_tag.uncorrected, tagBits)
    val tagMatch = s1_vb && tag === s1_tag
    s1_tag_disparity(i) := s1_vb && enc_tag.error
    s1_tl_error(i) := tagMatch && tl_error.asBool
    s1_tag_hit(i) := tagMatch || scratchpadHit
  }
```
ITIM should always hit if the accessing address locates  in the range of ITIM space, note that `scratchpadHit `uses io.s1_paddr(which is the paddr after TLB translation) to determine whether the address is in ITIM range. However, we have the following mismatch in data sram logic:
```
  for (((data_array, omSRAM), i) <- data_arrays zipWithIndex) {
    def wordMatch(addr: UInt) = addr.extract(log2Ceil(tl_out.d.bits.data.getWidth/8)-1, log2Ceil(wordBits/8)) === i
    def row(addr: UInt) = addr(untagBits-1, blockOffBits-log2Ceil(refillCycles))
    val s0_ren = (s0_valid && wordMatch(s0_vaddr)) || (s0_slaveValid && wordMatch(s0_slaveAddr))
    val wen = (refill_one_beat && !invalidated) || (s3_slaveValid && wordMatch(s1s3_slaveAddr))
    val mem_idx = Mux(refill_one_beat, (refill_idx << log2Ceil(refillCycles)) | refill_cnt,
                  Mux(s3_slaveValid, row(s1s3_slaveAddr),
                  Mux(s0_slaveValid, row(s0_slaveAddr),
                  row(s0_vaddr))))
    when (wen) {
      val data = Mux(s3_slaveValid, s1s3_slaveData, tl_out.d.bits.data(wordBits*(i+1)-1, wordBits*i))
      val way = Mux(s3_slaveValid, scratchpadWay(s1s3_slaveAddr), repl_way)
      data_array.write(mem_idx, Vec.fill(nWays)(dECC.encode(data)), (0 until nWays).map(way === _))
    }
    val dout = data_array.read(mem_idx, !wen && s0_ren)
    when (wordMatch(Mux(s1_slaveValid, s1s3_slaveAddr, io.s1_paddr))) {
      s1_dout := dout
    }
  }

```
When the access comes from the diplomatic ITIM port, it will use `s0_slaveAddr `to indicate the row address, however, ITIM  can also accept access from original io port, when the ITIM access comes from the original io, seems that it will use `s0_vaddr `to specify the read port address, I know the io port's default usage is for cpu cache access, and ICache is VIPT, so it is reasonable to use s0_vaddr to get the row address, but when the `io.req.addr ` locates in the address range for ITIM, seems it is a mismatch to use virtual address to specify the row address.  Note the last when clause above uses `io.s1_paddr`:
```
  when (wordMatch(Mux(s1_slaveValid, s1s3_slaveAddr, io.s1_paddr))) {
      s1_dout := dout
    }
```
Is this an issue that needs to take concern?
@aswaterman  @hcook

aswaterman, 2021-09-08:
It's correct.

jerryhethatday, 2021-09-09:
> It's correct.

Thanks for you clarification, seems that when VM and ITIM are both enabled, pgIdxBits >= untagBits must hold:
```
  require(!usingVM || outer.icacheParams.itimAddr.isEmpty || pgIdxBits >= untagBits,
    s"When VM and ITIM are enabled, I$$ set size must not exceed ${1<<(pgIdxBits-10)} KiB; got ${(outer.size/nWays)>>10} KiB")

```
Then the` row(s0_vaddr)` and  row(s1_paddr)` will be the same. 

Sorry for my inattention, and I have another problem, why you organize the data sram this way instead of the more reasonable way based bank like tag sram? Is that purely just for sending only the data the cpu needs instead of an entire row? Thanks again

closed 2021-10-18 by jerryhethatday